morrownr/rtw89 #1: rtw89_8922ae 0000:08:00.0: Direct firmware load for rtw89/rtw8922a_fw.bin failed with error -2

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
[image]

你在安装我的 rtw89 驱动前，先安装了 lwfinger 的 rtw89 对吧？因为我的 rtw89 驱动并不包含 rtw89pci 和 rtw89core 这两个模块
你没删除旧的 rtw89 就直接安装我的 rtw89，就会引起类似这样的怪问题，我会建议你先执行這些指令删除 lwfinger 的 rtw89
```
cd /lib/modules/$(uname -r)/kernel/drivers/net/wireless/realtek/rtw89
sudo rm rtw_* 
sudo rm rtw89core* 
sudo rm rtw89pci*
sudo depmod -a
```

删除了 lwfinger 的 rtw89 后重启动再尝试
正常来说 `lsmod | grep rtw` 应该只会看到 rtw_8922ae rtw_8922a rtw89 这三个 modules

a5a5aa555oo, 2025-02-10:
不对啊兄弟，你用的不是我的驱动啊
你下载到 lwfinger 的驱动了
我的驱动里不会 rtw89core.c rtw89pci.c 这个两个档案

huajian628, 2025-02-10:
我用你的驱动安装后还是没有 后加的那些

huajian628, 2025-02-10:
可以qq或者微信吗

a5a5aa555oo, 2025-02-10:
兄弟，你执行 `lsmod | grep rtw` 有看到 rtw_8922ae rtw8922a 和 rtw89 这三个 modules 吗？
有这三个 module 才代表你的系统确实载入了我的 rtw89

PS: 我没有 QQ 或微信

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
兄弟你要先刪除 lwfinger 的 rtw89 啊
你的系統還是載入了 lwfinger 的驅動 ...

我会建议你直接把 /lib/modules/6.8.0-53-generic/kernel/drivers/net/wireless/realtek/rtw89 整个删除
```
sudo rm -rf  /lib/modules/6.8.0-53-generic/kernel/drivers/net/wireless/realtek/rtw89
```

然後重裝我的驅動
```
sudo apt install git

git clone https://github.com/a5a5aa555oo/rtw89
cd rtw89
make clean modules
sudo make install
sudo make install_fw
```

之后重开机看看 rtw_8922ae rtw8922a rtw89 这三个 module 有没有载入

huajian628, 2025-02-10:
按照您说的操作了 但还是没有WiFi

a5a5aa555oo, 2025-02-10:
执行一下这两个指令
`modinfo rtw89core`
`modinfo rtw89`

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
你的系统还是只有 lwfinger 的 rtw89 驱动
没有我的驱动

你可以把 https://github.com/a5a5aa555oo/rtw89/issues/1#issuecomment-2647174429 的整个执行过程贴上来吗

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
你载入 rtw_8922ae 试试
`sudo modprobe rtw_8922ae`

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
执行一下 `sudo dmesg | grep 8922` 看一下有什么错误讯息

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
你执行这个指令然后把 dmesg.txt 这个档案附上来吧，谢谢

`sudo dmesg > dmesg.txt`

huajian628, 2025-02-10:
[dmesg.txt](https://github.com/user-attachments/files/18730697/dmesg.txt)

a5a5aa555oo, 2025-02-10:
你重開機然後再執行一次 `sudo dmesg > dmesg1.txt` 
然後再把 dmesg1.txt 附上來

huajian628, 2025-02-10:
[2.txt](https://github.com/user-attachments/files/18730848/2.txt)

a5a5aa555oo, 2025-02-10:
太奇怪了，你的系统怎么一直载入 rtw89pci
执行一下这个指令 `modinfo rtw89pci`
看一下这个档案到底在哪里

huajian628, 2025-02-10:
[image]

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
又是 lwfinger 的 rtw89驱动..
你执行这两个指令移除它們，然后重开机试试
如果还是不行，再把 dmesg 贴上来

```
sudo rm -v /lib/modules/6.8.0-53-generic/kernel/net/wireless/rtw* 
sudo depmod -a
```

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
重开机试试吧，不行的话再贴 dmesg.txt

huajian628, 2025-02-10:
[image]

a5a5aa555oo, 2025-02-10:
```
[    4.653532] rtw89_8922ae 0000:08:00.0: Unknown firmware header version 10
[    4.653536] rtw89_8922ae 0000:08:00.0: failed to recognize firmware
[    4.653537] rtw89_8922ae 0000:08:00.0: failed to setup chip information
[    4.653884] rtw89_8922ae: probe of 0000:08:00.0 failed with error -2
```
真難倒我了 ，rtw89 驅動正確載入了，但不曉得為什麼認不得 firmware

這樣吧，Linux kernel 自 6.10 開始支持 RTL8922AE
所以你可以試試把 kernel 升級到 6.10 以上
然後直接使用 kernel 裡的 rtw89 驅動 
**不要安裝我的 rtw89 驅動，也不要安裝 lwfinger 的驅動**

你可以安裝 mainline
```
sudo add-apt-repository ppa:cappelikan/ppa
sudo apt update
sudo apt install mainline
```
然後執行 `mainline-gtk`，接著安裝 kernel 6.13
以 kernel 6.13 開機，然後直接使用 kernel 內建的 rtw89 驅動試試

[image]

huajian628, 2025-02-10:
好的 我换另外一台机器测试下

huajian628, 2025-02-10:
这样升级内核对系统有什么影响吗

a5a5aa555oo, 2025-02-10:
6.8升到6.13应该不会有影响，这几年内核API的波动实际利率
像我的系统装了 kernel 5.4 ~ kernel 6.13 都没什么问题

huajian628, 2025-02-11:
[image]

huajian628, 2025-02-11:
[image]

huajian628, 2025-02-11:
升级内核我就没有成功过

a5a5aa555oo, 2025-02-11:
兄弟 抱歉 我也没办法啦
只能请你把 Ubuntu 升级到最新版试试
下面这里有 Ubuntu 25.04 的 LiveCD 
https://cdimage.ubuntu.com/daily-live/current/
Ubuntu 25.04 的 kernel 版号为 6.12
你试试看 RTL8922AE 能不能在 Ubuntu 25.04 下运作
如果不行 那就再用 mainline-gtk 把 kernel 升级到 6.13 试试
Ubuntu 25.04 提供的 libc6 很新，所以应该可以顺利升级到 6.13

a5a5aa555oo, 2025-02-11:
兄弟能否帮我个忙？
可以帮我看一下你的 /lib/firmware/rtw89/rtw8922a_fw-3.bin 的 SHA-1 是多少吗？
`sha1sum /lib/firmware/rtw89/rtw8922a_fw-3.bin`

spectrbro, 2025-02-11:
Whenever I download something, I am spammed with the following output from dmesg ```[  148.082217] rtw89_8922ae 0000:0f:00.0: failed to update 153 RXBD info: -11
[  148.082244] rtw89_8922ae 0000:0f:00.0: failed to update 154 RXBD info: -11
[  148.130228] rtw89_8922ae 0000:0f:00.0: failed to update 158 RXBD info: -11
[  148.130254] rtw89_8922ae 0000:0f:00.0: failed to update 159 RXBD info: -11
[  148.227153] rtw89_8922ae 0000:0f:00.0: failed to update 115 RXBD info: -11
[  148.227180] rtw89_8922ae 0000:0f:00.0: failed to update 116 RXBD info: -11
[  148.303535] rtw89_8922ae 0000:0f:00.0: failed to update 11 RXBD info: -11
[  148.303908] rtw89_8922ae 0000:0f:00.0: failed to update 12 RXBD info: -11
[  148.338586] rtw89_8922ae 0000:0f:00.0: failed to update 107 RXBD info: -11
[  148.338641] rtw89_8922ae 0000:0f:00.0: failed to update 108 RXBD info: -11
[  148.369723] rtw89_8922ae 0000:0f:00.0: failed to update 166 RXBD info: -11
[  148.369767] rtw89_8922ae 0000:0f:00.0: failed to update 167 RXBD info: -11
[  148.605068] rtw89_8922ae 0000:0f:00.0: failed to update 6 RXBD info: -11
[  148.605095] rtw89_8922ae 0000:0f:00.0: failed to update 7 RXBD info: -11
[  148.669703] rtw89_8922ae 0000:0f:00.0: failed to update 154 RXBD info: -11
[  148.669730] rtw89_8922ae 0000:0f:00.0: failed to update 155 RXBD info: -11
[  148.859018] rtw89_8922ae 0000:0f:00.0: failed to update 135 RXBD info: -11
[  148.859044] rtw89_8922ae 0000:0f:00.0: failed to update 136 RXBD info: -11
[  148.864954] rtw89_8922ae 0000:0f:00.0: failed to update 212 RXBD info: -11
[  148.864981] rtw89_8922ae 0000:0f:00.0: failed to update 213 RXBD info: -11
```

From what I can gather it is a problem with how DMA interacts with the RX buffer. 

https://patchwork.kernel.org/project/linux-wireless/patch/20240425112816.26431-3-pkshih@realtek.com/

The error is generated in this file:

https://github.com/lwfinger/rtw89/blob/main/pci.c

In rtw89_pci_rxbd_deliver_skbs

‘

PCI RX ring is a kind of read/write index ring, and DMA and ring index are
asynchronous, so suddenly driver gets newer index ahead before DMA. To
resolve this rare situation, we use a RX tag as helpers to make sure DMA
is done.

The RX tag is a 13-bit value, and range is from 1 ~ 0x1FFF, but 0 isn't
used so should be skipped.

Only enable this validation to coming WiFi 7 chips, because existing
chips use different design and don't really meet this situation.

Add missed rx_ring_eq_is_full for 8851BE by the way.
‘

a5a5aa555oo, 2025-02-12:
Hi, @spectrbro 

I wonder what distro do you use, Could you please attach the full dmesg as well? thanks.

`sudo dmesg > dmesg.txt`

huajian628, 2025-02-12:
您好 我尝试安装了Ubuntu24.04.1桌面版 还是没有WiFi驱动 你说的镜像 在其他站点好像找不到  我下载了  等下可以安装试试 还要我验证什么吗 最好的结果就是可以在Ubuntu22.04 24.04正常可用WiFi  不知道这个是否可以制作backport

a5a5aa555oo, 2025-02-12:
Ubuntu 24.04 的 kernel 为 6.8，一样不支持 RTL8922AE
我帮你测试一下 Ubuntu 24.04 能不能装上 kernel 6.13

a5a5aa555oo, 2025-02-12:
理论上 6.10 以上的 kernel 都支持 RTL8922AE
但实际上要试了才知道
我只有RTL8852BE，没有 RTL8922AE 可以测试

我会建议你找一些包含最新 kernel 的 LiveCD 做测试
像是 EndeavourOS
EndeavourOS 2025.02.08 版用的 kernel 版本为 6.13.1
你可以在 LiveCD 环境底下测试你的 RTL8922AE 能不能运作 (不需要将 EndeavourOS 安装到你的电脑)
如果可以，你再把 Ubuntu 24.04 的 kernel 升级到 6.13

下载连结：https://endeavouros.com/

a5a5aa555oo, 2025-02-12:
兄弟我帮你测试完了 Ubuntu 24.04 的 kernel 可以顺利升级到 6.13

[image]
[image]

huajian628, 2025-02-12:
那我试试

huajian628, 2025-02-12:
不用升级  我刚才安装了Ubuntu24.04.1 然后在系统里面更新 sudo apt dist-upgrade -y  重启后系统内核会更新到6.11  WiFi可用了

huajian628, 2025-02-12:
[image]

huajian628, 2025-02-12:
网口4是我插的USB无线网卡

a5a5aa555oo, 2025-02-12:
[image]

huajian628, 2025-02-12:
那我是否可以安装Ubuntu22.04然后再升级内核到6.11试试?

huajian628, 2025-02-12:
大佬 您是否可以制作backport dkms deb包?

huajian628, 2025-02-12:
直接在低版本内核系统安装dkms deb驱动包就好了

huajian628, 2025-02-12:
[image]

huajian628, 2025-02-12:
新的问题 蓝牙不可用 打不开

huajian628, 2025-02-12:
f**k the rtw

a5a5aa555oo, 2025-02-12:
> 那我是否可以安装Ubuntu22.04然后再升级内核到6.11试试?

我刚试了，Ubuntu 22.04 的 kernel 没办法透过 mainline-gtk 升级到 6.10 / 6.11 / 6.12 / 6.13

> 大佬 您是否可以制作backport dkms deb包?
我没做过 deb 包，可能要先去学习学习

huajian628, 2025-02-12:
我可以提供环境给您测试

[54 more replies]