morrownr/rtw89 #3: rtw89_8852be 0000:02:00.0: AMD-Vi: IO_PAGE_FAULT

Metztli, 2025-04-04:
Hello-

For over a month ago till now, I have been using your rtw89 for the kernel below:

Linux version 5.17.0-3+reizer4.0.2-amd64 (debian-kernel@lists.debian.org) (gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40) #1 SMP PREEMPT Debian 5.17.13-2+reizer4.0.2 (2025-02-11)
...

To be able to install Debian from the network, I hacked your modules into my kernel build so that the Debian Installer (d-i) could fetch the software components via WiFi. It worked like magic. Thanks!

Notwithstanding, sporadically I notice the below entries when viewing the dmesg log:

[37544.490459] rtw89_8852be 0000:02:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0x0 flags=0x0000]
...
[48447.426051] rtw89_8852be 0000:02:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0x0 flags=0x0000]

It seems the quality of the connection decreases when that happens. Any insight and/or suggestion will be appreciated.

Thank you again for your work.

Metztli, 2025-04-07:
As a matter of fact, today I was using VirtualBox 7.1.6 and suddenly lost Internet connection; dmesg output showed, after one of those lines referenced above, I got a rtw89_8852be induced kernel crash:

[36124.620212] rtw89_8852be 0000:02:00.0: AMD-Vi: Event logged [IO_PAGE_FAULT domain=0x000a address=0x0 flags=0x0000]
[36124.672087] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36126.688056] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36128.960036] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36130.784055] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36132.608093] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36134.848058] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36138.943748] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36140.735796] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36142.751881] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36144.735801] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36150.979833] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36153.791833] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36155.744225] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36157.887957] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36159.776459] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36161.792057] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36163.776232] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36165.696283] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36167.683420] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36169.695950] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36171.680086] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36173.696279] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36175.680389] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36177.632740] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36179.651949] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36181.695957] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36183.742953] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36185.696001] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36187.651147] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36187.653601] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36187.653605] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.580139] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.580154] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.581107] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.581110] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.632026] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36189.632031] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.632032] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.632295] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.632297] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.634445] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.634446] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.634454] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.634455] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.634459] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.634459] rtw89_8852be 0000:02:00.0: failed to send h2c
[36189.634506] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36189.634507] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.564168] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.564183] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.565145] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.565148] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.616076] rtw89_8852be 0000:02:00.0: rtw89: failed to leave lps state
[36191.616080] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.616081] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.616332] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.616333] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.618479] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.618480] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.618488] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.618490] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.618493] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.618494] rtw89_8852be 0000:02:00.0: failed to send h2c
[36191.618540] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36191.618541] rtw89_8852be 0000:02:00.0: failed to send h2c
[36193.676688] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36193.676699] rtw89_8852be 0000:02:00.0: failed to send h2c
[... 1533 lines cut ...]
[36349.399387] rtw89_8852be 0000:02:00.0: failed to send h2c
[36349.399394] rtw89_8852be 0000:02:00.0: no tx fwcmd resource
[36349.399394] rtw89_8852be 0000:02:00.0: failed to send h2c
[36362.855444] wlp2s0: authenticate with 82:qr:st:wx:yz:28
[36363.036620] wlp2s0: send auth to 82:qr:st:wx:yz:28 (try 1/3)
[36363.037334] wlp2s0: authenticated
[36363.044801] wlp2s0: associate with 82:qr:st:wx:yz:28 (try 1/3)
[36363.088804] wlp2s0: RX AssocResp from 82:qr:st:wx:yz:28 (capab=0x1131 status=0 aid=9)
[36363.205013] wlp2s0: associated
[... cut ...]
[36451.560116] wlp2s0: authenticated
[36451.561120] wlp2s0: associate with 82:qr:st:wx:yz:28 (try 1/3)
[36451.587103] wlp2s0: RX AssocResp from 82:qr:st:wx:yz:28 (capab=0x1131 status=0 aid=8)
[36451.705320] wlp2s0: associated
[36451.705340] IPv6: ADDRCONF(NETDEV_CHANGE): wlp2s0: link becomes ready
[36451.705606] wlp2s0: Limiting TX power to 30 (30 - 0) dBm as advertised by 82:qr:st:wx:yz:28
[36677.624126] sysrq: Show Blocked State

I had to reboot since I was unable to bring back the WiFi device.

Thanks.

a5a5aa555oo, 2025-04-17:
Hi @Metztli,

A [bug report](https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/) said this issue may be caused by the firmware, maybe you can try degrading the firmware for 8852BE to v0.29.29.5.

Steps to degrade the firmware:
`git clone -b 6.6-lts https://github.com/a5a5aa555oo/rtw89`
`cd rtw89`
`sudo make install_fw`

Then reload the driver. (or reboot)
`sudo rmmod rtw_8852be rtw_8852b rtw89`
`sudo modprobe rtw_8852be`

Metztli, 2025-04-17:
> A [bug report](https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/) said this issue may be caused by the firmware, maybe you can try degrading the firmware for 8852BE to v0.29.29.5.

Hello @a5a5aa555oo,

Thanks for your reply and suggestion; however, v0.29.29.5  is what Debian has been already loading for 8852BE ever since I noticed the issue ;-)

i.e.,
[image]

Metztli, 2025-05-19:
Hello @a5a5aa555oo,
> 
> A [bug report](https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/)  ...
> 
I have been monitoring the thread you referenced and, earlier, I have upgraded the rtw8852b_fw-1.bin firmware. Thus now, I am running v1.29.29.9

[image]

I have not noticed any similar issue as that which opened this thread; notwithstanding, I will continue monitoring firmware stability.

Best regards.

Metztli, 2025-05-29:
Hello @a5a5aa555oo,

After running for a while Firmware version 1.29.29.9 referenced above, i.e.,
rtw89_8852be 0000:02:00.0: Firmware version 1.29.29.9 (4a4fba35), cmd version 0, type 5
rtw89_8852be 0000:02:00.0: Firmware version 1.29.29.9 (4a4fba35), cmd version 0, type 3

I have not seen IO_PAGE_FAULT anymore. Thus, closing issue now.

Again, thank you for your work!

P.S. I have since integrated the rtw89 firmware 1.29.29.9 and your module into [Metztli Reiser4](https://sourceforge.net/projects/metztli-reiser4/) enhanced kernel build 5.17.13-2 for Intel and AMD.

a5a5aa555oo, 2025-06-14:
@Metztli 

Thank you for the report and glad the problem was solved!

Currently the latest version of 8852B firmware in linux-firmware repo is still [0.29.29.8](https://web.git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/log/rtw89/rtw8852b_fw-1.bin). I don't know why firmware v1.29.29.9 are not sent to linux-firmware, maybe that firmware is still under testing or has some potential problem.

I will update the 8852B firmware in this repo once Realtek developers send a new firmware to linux-firmware, thank you!