morrownr/rtw89 #4: Development Discussion (Pinned)

morrownr, 2025-05-14:
This issue is being established as a location for developers to discuss issues related to the future direction of this repo.

Initial USB support recently went into linux-wireless to be included in mainline rtw89. This is a significant step toward having proper mac80211 driver support for Realtek WiFi 6 and 7 USB adapters. Bringing USB adapter/module support to rtw89 will take a LOT of work and will require a lot of testing. No timeline has been established at this point

Hopefully this issue can serve as a fixed location to discuss development issues of interest.

@morrownr

morrownr, 2025-05-14:
@dubhater 

I see that you have merged several patches lately. Can you help me and anyone else that stops by to understand the current status  and your plan moving forward?

My plan for now is to make your plan available on my USB-WiFi repo in an effort to gain additional help to maintain this repo. Right now I only have 2 Realtek WiFi 6/7 adapters:

rtl8832bu
rtl8832cu

What do you need? And what do I need?

dubhater, 2025-05-15:
@morrownr RTL8851BU and RTL8852BU drivers are ready for testing.

The driver could still have problems when the device is unplugged or when the driver is unloaded. These problems should be reported.

The upload speed is not as good as it should be. Something to investigate later.

There is no switching to USB 3 yet.

When these first patches are accepted into the kernel, I will send the patches for RTL8852BU. After that, maybe USB 3 switching.

I also got RTL8832CU to work but something is wrong. It keeps losing the connection.

morrownr, 2025-05-18:
@dubhater 

I was really sick for two days after my last message. I hope I can get fully up to speed soon.

> I also got RTL8832CU to work but something is wrong. It keeps losing the connection.

I have not installed this driver yet but need to do so soon to familiarize myself. Do you want me to test and report on the rtl8832cu?

You did not mention the rtl8852/32 chip nor the rtl8912au chip. Do you have plans for them?

I understand the rest of what you said. Let know what you need.

morrownr, 2025-05-18:
@a5a5aa555oo 

> Kernel 6.6 ~ 6.14 are supported. For kernel 5.15 ~ 6.5 users, use the rtw89 driver in 6.6-lts branch instead.

I noticed that note in the readme. I had previously helped a user backport the mt7921u driver to 5.15. It was reasonable to backport the mt7921u driver from kernel 6.1 but 6.6 or above was basically an insurmountable challenge due to the numerous changes in the supporting wifi stack between 6.1 and 6.6 so I assume you ran into similar issues.

Since this is going to be a repo used for wide testing, my suggestion is that we remove the above statement from the readme as it will cause problems... especially when users are testing rtl8832cu and rtl8912au chips. Can I remove that statement?

I am slowly familiarizing myself with the repo. You have done an incredible amount of work. What repo did you use to keep the code synchronized and how did you go about doing so? Larry was using wireless-next but he passed away before I had the chance to ask him how he went about doing it.

morrownr, 2025-05-18:
@a5a5aa555oo 

> Please don't remove it because the driver in main branch works for kernel 6.6+ only.

I would argue that rtw89 should support only kernel 6.6+. We may have differences in the vision we have but I'm sure good discussion can lead to compromise.

> I would be very grateful if you could backport it to kernel 5.15. ^_^

That is not going to happen. I have already been down this road with mt76. Trying to backport this or any other wifi driver from kernel 6.6+ back to kernels 5.15 to 6.1 is a bridge too far. There were simply too many stack changes during year 2023 (kernel 6.1 to 6.6). The changes are such that it is leads to nearly impossible situations, short of backporting mac80211 and requiring users to install it but then that can lead to other problems. OpenWRT skipped kernel 6.1 and went to kernel 6.6 for this latest version because they know. If this repo tries to support anything older than kernel 6.6, the cost will be very high. IF you want to read a novel I can did out more details.

> Oh, sorry! now i know what you mean... Orz
Could you please tell the users to install the kernel 6.6 or later for testing?

Maybe a reasonable compromise if for me to change the line to:

Note: Testers should the default branch `main`. Kernel 6.6+ is required.

I understand about users whining about wanting things backported. However, my experience is that this is mostly limited to owners and users or SBC's that are from companies that do a poor job of supporting their products. The only company in the SBC sector that does this right is Raspberry Pi. The recent update to RasPiOS uses kernel 6.12 and it works on every single RasPi hardware product that has ever been released. Nvidia is very guilty of stranding users. Radxa is also guilty of this. My position is that the cost of trying to support problems caused by bad decisions of various companies is too high of a cost to our mission here. Our mission to code, test and upstream great quality USB drivers for the Realtek rtw89 driver series. We have things to overcome that are not even on the radar yet. Unlike the Mediatek WiFi 6 and 7 adapters, the vast majority of these Realtek WiFi 6 and 7 adapters are multistate. This has shown, historically, to be problematic. I know first hand the adapters based on the rtl8852/32cu are problematic on some hardware. I learning this first hand here on my own systems and from testers as I was bringing the vendor driver online. There has to be a limit to support. If it was my repo and my decision, I would delete the branch that is backported to 5.15 and be done with it. 

Another thing I would do to support the mission here is to take dkms instructions out of he README. Manual basic installation will help force testers upgrade so that we are getting more accurate test results. What I can do is make a reduced function set of installation scripts that only support an automated basic install plus secure boot capability. That would automate and standardize the installation and upgrade process allowing for less setup and maintenance on the part of testers so that they can concentrate of what we want them to concentrate on. The script could also add information in a file that could be included in any report so as to reduce the back and forth of question asking. The new script won't happen this week but I am making some progress finding help with my USB-WiFi site. Another issue is that we should expect testers to show up slower and will less skill on average. With rtw88, I was able to channel users of RTW88 supported adapters fairly rapidly because I had run vendors drivers for WiFi 5 chips for years and even had relationships with some of the very knowledgeable users.  I have none of that with WiFi 6 and no vendor driver for WiFi 7 exists to my knowledge.  The more knowledgeable Linux users have been migrating to Mediatek based adapters for several years now given the continued shitty USB support from Realtek's USB driver group so the best we can expect on average is users that have bought Realtek adapters because they did not know better. Anything we can do to make installation and maintenance easier on them is a good thing and providing information for reports will help greatly.

> I use wireless-next, too.

I will add a note about this in README.

> just merge the patches one by one, do the compilation test against every kernel version, then fix the errors.

Are you just looking at wireless-next to determine the applicable patches? And where are you in that process?

a5a5aa555oo, 2025-05-19:
@morrownr 

What do you think about that I transfer the rtw89 driver in main branch to you so that you can do anything without my permission?
you should stop maintaining those buggy vendor driver from Realtek and spend more time on this driver.
on the other side, i will keep maintaining the rtw89 driver in 6.6-lts branch which I use every day.

> Are you just looking at wireless-next to determine the applicable patches? 

I just merge [all of them](https://web.git.kernel.org/pub/scm/linux/kernel/git/wireless/wireless-next.git/log/drivers/net/wireless/realtek/rtw89)...

> And where are you in that process?

I dont know what you mean, sorry for my poor English...

morrownr, 2025-05-19:
> What do you think about that I transfer the rtw89 driver in main branch to you so that you can do anything without my permission?

I agree and I will leave the branch alone.

> you should stop maintaining those buggy vendor driver from Realtek and spend more time on this driver.

I agree and I have stopped maintaining the WiFi 5 drivers effective as of kernel 6.14, however, my site is far more than some buggy Realtek drivers. It is a hub of information and with worldwide users averaging about 25,000 hits per week, it generates a lot of message traffic, of which, I get all of it. It isn't just a few end-users. Mediatek devs use it to test patches. I am working to get some help.

> I just merge all of them...

> I don't know what you mean, sorry for my poor English...

Your English is excellent. Mine probably needs work. What I was asking is which fix is the last one that you have applied so that I know where to start?

I do keep a local copy of wireless-next here. I updated it last night and compared it to your rtw89 and I see, with meld, very few differences other that the necessary differences to allow for API changes so your repo is very current.

Could you go through the steps you use to take a patch from your link above to getting it into your rtw89? I am not used to pulling in patches from a repo such as wireless-next.

Regards

a5a5aa555oo, 2025-05-19:
> I agree and I will leave the branch alone.

I've deleted 6.6-lts branch.

> Your English is excellent. Mine probably needs work. What I was asking is which fix is the last one that you have applied so that I know where to start?

the last one is [wifi: rtw89: fix firmware scan delay unit for WiFi 6 chips](https://github.com/a5a5aa555oo/rtw89/commit/8346ed89c2361152b286100e710473c9f996606c)
currently this repo is fully synchronized with wireless-next.

> Could you go through the steps you use to take a patch from your link above to getting it into your rtw89? I am not used to pulling in patches from a repo such as wireless-next.

I will give you detailed steps tomorrow.

dubhater, 2025-05-20:
@morrownr 

> I have not installed this driver yet but need to do so soon to familiarize myself. Do you want me to test and report on the rtl8832cu?

That chip can't be tested yet. It needs more changes in usb.c and elsewhere just to get it working badly.

> You did not mention the rtl8852/32 chip nor the rtl8912au chip. Do you have plans for them?

I would like to have plans for them but I don't have those chips.

@a5a5aa555oo 

> Do you need kernel 6.6? I recompiled kernel 6.6.88 yesterday because 6.6.89 and 6.6.90 are buggy, download from here if you don't want to build by yourself.

I don't need it right now, but thank you. It takes about 1 hour to compile a stripped down kernel on my AMD Athlon Gold.

morrownr, 2025-05-20:
@dubhater 

> The rtl8832cu chip can't be tested yet. It needs more changes in usb.c and elsewhere just to get it working badly.

Copy that. Standby mode for rtl8852/32cu.

> I would like to have plans for the rtl8852/32au and rtl8912au chips but I don't have those chips.

Let me come up with a plan. I need adapters based on the rtl8851bu and rtl8912au myself.

As far as I can tell, there is absolutely no Linux driver for the rtl8912au chip. Is that how you see it right now?

dubhater, 2025-05-20:
I think there is no published Linux driver. If it exists but it's not published yet, Ping-Ke might be able to help me.

morrownr, 2025-05-20:
@dubhater 

> Ping-Ke might be able to help me.

You can always ask. As long as you are asking, I do not have good sources for current versions of the rtl8851bu or rtl8852au vendor drivers.

The rtl8912au is a WiFi 7 chip so the driver will kick in the kernel code to shut down WEXT APIs. Maybe the Realtek USB folks are struggling with that.

luksab, 2025-05-22:
> I need adapters based on the rtl8851bu and rtl8912au myself.

I have an RTL8851BU and an RTL8852BU based usb each that I could send over if that would be of any help. If so, feel free to reach out to me over the email on my profile.

morrownr, 2025-05-22:
FYI: Dangerous storms passed through my area over two days earlier this week. Our home sustained some damage but not major damage. I was without primary internet access for 3 days. My wife and I are okay but were previously sick and are still taking medications to recover. I was able to use my phone to check in. It may be awhile before I am fully up to speed.

@dubhater @a5a5aa555oo 

I was able to post a message in USB-WiFi issues today regarding this project. I have prioritized getting the needed hardware in the hands of those who need it. I did not mention you, @a5a5aa555oo , as I do not know what you need. You are part of this team so if you want me to facilitate getting hardware to you, let me know what you need.

I am testing this repo with a rtl8832bu adapter. So far, nothing to mention. It works.

a5a5aa555oo, 2025-05-22:
@morrownr 

Thank you, but it's a kind of waste to send me the adapters as im unable to write a driver.

i have an 8852BE and i think it's enough, i can use it to test this driver.

morrownr, 2025-05-23:
@a5a5aa555oo 

> please consider not deleting DKMS instructions in README.md, some testers who are unable to disable Secure Boot may need them

Agree.

morrownr, 2025-05-23:
@dubhater 

You had mentioned issues with the rtl8852cu driver and I happened to be going over the firmware files. The rtw8852c_fw-1.bin file here in this repo is not the same as in the rtw89 storage location:

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw89

```
$ ls -l *.bin
-rw-r--r-- 1 morrow morrow 2375560 May 23 01:01 'rtw8852c_fw-1 (1).bin'
-rw-r--r-- 1 morrow morrow 2399272 May 23 00:55  rtw8852c_fw-1.bin
```

Do you think I should go ahead and merge what appears to be an updated version?

It is not clear to me why there are 3 versions of the rtw8852c files. Any idea?

dubhater, 2025-05-23:
We have rtw8852c_fw.bin, rtw8852c_fw-1.bin, and rtw8852c_fw-2.bin because they made some changes which break compatibility with older kernels. Old kernels will use rtw8852c_fw.bin. Theoretically, 6.12 and newer kernels can use rtw8852c_fw-1.bin or rtw8852c_fw.bin, and 6.16+ can use all three files.

It should be fine to update them.

morrownr, 2025-05-23:
@dubhater @a5a5aa555oo 

Okay, that explains a few things. There are two other rtw89 chips that have multiple firmware files that make it look like that they ended up in the same situation:

rtw8852b
rtw8922a

I think the only reasonable solution at this point is to add all of the firmware files to this repo in their latest form. Let me know if you see any potential problems with this approach.

For what it is worth: This actually does not surprise me at all. This falls in line with the discussion @a5a5aa555oo were having regarding the backporting of this repo all the way back to kernel 5.15... remember that I said that won't happen. It appears that Realtek ran into the same things I ran into trying to backport mt76. There were changes in the APIs during 2023 (kernel 6.1 to 6.6) that made for some nearly impossible situations. That led to 2024 having a lot of wireless problems. Hopefully things can stabilize greatly over the course of this year.

Distro maintainers likely will not be happy about the wasted space that is caused by having multiple very large firmware files for the same chip. I made bug repo to RasPiOS last week and one of the things that I reported was the very dated Mediatek wifi firmware files. The mt7921 firmware files were at least 5 versions old. This is not unique to RasPiOS. The only distro that I know of that does a reasonably decent job of keeping firmware up to date is Ubuntu. Distro maintainers really need to have a script that runs maybe every 2 weeks to pull new and updated firmware so the updates can then be flowed out to users in a timely manner.

Anyway, I'll plan to make the firmware folder here in this repo match the kernel Realtek firmware rtw89 folder unless you guys see a problem.

a5a5aa555oo, 2025-05-23:
@morrownr 

Old firmwares are useless for this driver, you only need to add the latest firmware images.

morrownr, 2025-05-23:
> ... you only need to add the latest firmware images.

What we need to do is see what filename the individual drivers are looking for. Given the situation that @dubhater outlined, the drivers may, or should, have some `if then` situations based on what kernel is in use. I have already synchronized the firmware folder based on what I know right now but I can revert or make other changes as this becomes more clear. Well, I thought it was clear 15 minutes ago and now I am not so sure.

@dubhater 

Here is a quick and incomplete look at the firmware filenames in some of the drivers:

$ grep -r "rtw8852b_fw" *
rtw8852b.c:#define RTW8852B_FW_BASENAME "rtw89/rtw8852b_fw"

$ grep -r "rtw8852c_fw" *
rtw8852c.c:#define RTW8852C_FW_BASENAME "rtw89/rtw8852c_fw"

$ grep -r "rtw8922a_fw" *
rtw8922a.c:#define RTW8922A_FW_BASENAME "rtw89/rtw8922a_fw"

It is not clear to me how the various firmware files for various kernels are being handled. Can you clarify?

morrownr, 2025-05-23:
> rtw89 tries to load the latest firmware (rtw8852c_fw-2.bin). If it's not available, it tries the next one (rtw8852c_fw-1.bin). If that's not available, then the next one (rtw8852c_fw.bin). And so on. It's the same for the other chips.

Good explanation. That is not a way of handling firmware that I had considered. That may confuse the hell out of many distro maintainers.

Given my newfound understanding, I have the option to delete the older files for each chip and I will do so soon. Sorry for the confusion.

a5a5aa555oo, 2025-05-24:
@morrownr 

I've deleted those unused firmware files

The firmware v0.29.29.8 for 8852B has some problems, so downgrade it to v0.29.29.5 also.

https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/

morrownr, 2025-05-24:
@a5a5aa555oo 

Thanks. I understand about the firmware bug. Thanks for the link. 

I have been pondering the firmware situation. I started USB-WiFi to help the Linux USB WiFi situation as it was not so good some 8+ years going back. It has improved greatly in the last 8 years. I'd say the biggest factors in the overall satisfaction level of USB WiFi adapters and modules has been these factors:

1. Mediatek's decision a little over 8 years to get involved with linux-wireless and produce mac80211 drivers.
2. dubhater's role in bringing usb support and drivers to rtw88.
3. Pink-Ke's willingness to assist in adding USB support to rtw88 and rtw89.
4. The information that is available at USB-WiFi.

Progress continues but there are areas where progress is needed. I think firmware management is an area that needs a look. I am not saying that I have a solution yet but I see uneven results in how distros are handling wifi firmware. The failure of distros to update firmware when new files are released is problematic. Then the issue of the many maintainers that can't keep up with the various ways that firmware updates are handled is an issue. How would distros maintainers know this interesting little scheme that is being used with rtw89? Is there documentation that I don't know about?

Do we need a script that supports pulling Realtek and Mediatek firmware files that can also update a distro? It could be built in a way that full Linux firmware support could be added. That could help standardize and speed up the process used by distros and end users.

a5a5aa555oo, 2025-05-24:
@morrownr 

I dont know what to say,  the discussion below may tell you why there exists mutilple firmware files for specific chip and how developers handled the compatibility problem.
https://lore.kernel.org/linux-wireless/df1ce994-3368-a57e-7078-8bdcccf4a1fd@gmail.com/T/#maaedda3472cfb9bbbc5a10a58c5f268e3555bd0e

> Do we need a script that supports pulling Realtek and Mediatek firmware files that can also update a distro? It could be built in a > way that full Linux firmware support could be added. That could help standardize and speed up the process used by distros and > end users.

You cannot update the firmware files only on a system because "old driver + latest firmware" may cause problems sometimes.

morrownr, 2025-05-25:
> I dont know what to say, the discussion below may tell you why there exists multiple firmware files for specific chip and how developers handled the compatibility problem.

I remember that discussion. What that discussion tells me is that what we are seeing should NOT happen. Period. From Kalle in the discussion:

```
We cannot break user space like that, linux-firmware needs to be
backwards compatible. In other words, anything you submit to
linux-firmware needs to work with old kernels.
```

The more I think about this, I really think this issue needs to go upstream to linux-wireless leadership and/or to Greg and maybe to Linus. I remember something similar happened with Mediatek. They were told to fix it and fix it in a way that does not have the potential to break user space. It took a lot of work but they did it.

> You cannot update the firmware files only on a system because "old driver + latest firmware" may cause problems sometimes.

Oh, yes you can if the firmware is programmed like it should be. I do this regularly. What we have going on here is wrong. How the hell do we know what firmware goes with what kernel? And if we don't know, how the hell are the hundreds of distro maintainers going to know? There are long standing rules against this. Maybe you and/or dubhater could have a discussion with Ping-Ke. I can easily write a script for Mediatek firmware without any fear of running the latest with any kernel. Filenames stay the stay the same. I had no idea Realtek was doing this with their WiFi 6 firmware and it seems at least 3 different chips are involved at this point. Maybe you can convince Ping-Ke to do something about this. I will wait to see but at some point, if I do not see any indication that this is being worked, I will send a detailed report to various levels up to and including Linus.

morrownr, 2025-05-25:
>  And if we don't know, how the hell are the hundreds of distro maintainers going to know?

As an example of a problem that this multiple different firmware files can cause, I checked a freshly installed Ubuntu 25.04. I decided to do this because Ubuntu pretty much sets the gold standard as far as keeping its firmware current. When I checked in /lib/firmware/rtw89 , what I saw was everything in the firmware rtw89 repo. Everything! Ubuntu uses kernel 6.14. I'm sure the maintainer expects firmware to be in accordance with the firmware guidelines:

```
Firmware Guidelines

Users switching to a newer kernel should not have to install newer firmware files to keep their hardware working. At the same time updated firmware files must not cause any regressions for users of older kernel releases.
```

I suspect that there is not a single distro firmware maintainer that has a clue what is going on here with rtw89 and it is going to cause very unnecessary bloat in distros all across the Linux spectrum. 

I can appreciate some of the details that Realtek ran into regarding that kernel 6.1 to 6.6 period. I have ran into it myself trying to backport drivers from kernels 6.6+ to kernels 6.1- and like I said, it is not going to happen. In my case, it is not going to happen because I do not have the resources to tackle the problem so backporting 6.1 to 5.15 worked fine as this was not something to go into the kernel. What needs to happen is Realtek needs to suck it up and accommodate the problems in the driver, not the into the kernel or firmware repo. Mediatek did it, Realtek can do it also.

Lastly, my intention is not to create tension here. It is to get to the bottom of something that I did not understand because it goes against what I have understood are the rules for many years. Distro maintainers likely do not understand what is going on but once it is explained to them, there are going to be emails flying. 

I'll end this discussion for now and press on with learning this distro and testing the rtl8832bu driver.

dubhater, 2025-05-31:
I added a new error message to the USB code: "mutex already locked elsewhere". If you ever see this in dmesg, please let me know. (It's harmless.)

@morrownr 

> What is in the system to ensure that a change in the needed firmware file is accommodated when a kernel change takes place?

I don't know if they have something like that. But theoretically, newer rtw89 can use the older firmware.

> I will continue to work toward putting the 2 adapters that you need in your hands.

Thank you!

> I realize I speak about Ubuntu a lot but it is simply a testing distro here. I do not use Ubuntu as my daily driver as I do not like the gnome interface.

I don't like it either, I prefer KDE Plasma.

morrownr, 2025-06-01:
@dubhater 

What distro with KDE are you using? I try KDE around every 2 years. I like the interface but I keep ditching because of bugs. It has been a while. Should I try again?

Something that I am running into is figuring out a good rtl8851bu based adapter. It appears that this chip is a AX900 capable chip. AIC Semi also makes a chip that is AX900 capable so it can be hard to determine what adapter has what chip. Eventually I can find one but thought I'd see what you know.

dubhater, 2025-06-01:
I use Arch Linux, not the most convenient to install. Sure, you should try KDE again. Fresh bugs are introduced all the time. :)

The TP-Link TX10UB Nano should be fine. I have this generic one:

[image]

I think the AIC Semi chip supports a different Bluetooth version and the adapters I've seen are twice longer.

a5a5aa555oo, 2025-06-04:
@morrownr 

Could you take over [rtw89-legacy](https://github.com/a5a5aa555oo/rtw89-legacy)?
I think it would be better to put theses two rtw89 repo together.

morrownr, 2025-06-05:
@dubhater 

> I use Arch Linux, not the most convenient to install. Sure, you should try KDE again. Fresh bugs are introduced all the time. :)

I decided to go with Manjaro KDE for testing right now. One of the first things I discovered is my rtl8812au based adapter is not plug and play even though the installation of Majaro is using kernel 6.14 .x. I decided to investigate.

First I checked firmware and the firmware was in place like it should be. However, no driver was in the correct location. So, I'm wondering if this is because Arch has not turned on compiling of the driver or if Manjaro has messed this up on their own. Do you know if the driver is compiling on the most current Arch?

I have been seeing this on other distros. I am posting reports as I have time. I have to say that Ubuntu does a good job of turning on drivers and loading up the firmware, Debian and other distros... not so much.

morrownr, 2025-06-05:
@a5a5aa555oo 

> Could you take over rtw89-legacy? I think it would be better to put theses two rtw89 repo together.

Don't expect perfection as it is taking me time to get where I can spend more time with rtw89. Go ahead and do it.

Something that I am not comfortable with is that I am a single point of failure for too many things here at my site. I need to come up with a plan in case something happens to me. I am not young anymore.

dubhater, 2025-06-05:
@morrownr 

> So, I'm wondering if this is because Arch has not turned on compiling of the driver or if Manjaro has messed this up on their own. Do you know if the driver is compiling on the most current Arch?

Yes, the RTL8812AU and RTL8821AU drivers are enabled in Arch Linux since 6.13-arch1-1. Maybe Manjaro doesn't use the kernel from Arch Linux.

a5a5aa555oo, 2025-06-07:
> I also got RTL8832CU to work but something is wrong. It keeps losing the connection.

Someone's 8852CE keeps disconnecting too, maybe there is something wrong in the 8852C code?

https://github.com/lwfinger/rtw89/issues/388#issuecomment-2951213381

dubhater, 2025-06-07:
@a5a5aa555oo Sure, that looks like something to report to him.

dubhater, 2025-06-07:
@morrownr 

> Something that I am running into is figuring out a good rtl8851bu based adapter.

The Edimax EW-7611UXB is another option. According to the user manual, it should be "plug-and-play", but if it isn't for some reason, you have to download the Windows driver from their site, so it doesn't have the CD-ROM mode. Amazing!

It has a dual colour LED, blue and red. I have never seen a wifi adapter with a red LED. I wonder if you can see both colours at the same time. The EW-7611**UCB** has blue and green LEDs, but the blue is always on when the Bluetooth is on and it completely swamps out the green.

dubhater, 2025-06-08:
Oh, it's a new product, only announced in January.

The repository I use is already updated to the version from Edimax: https://github.com/fofajardo/rtl8851bu

We'll see how it goes with the patches. I hope to finish v2 today.

natimerry, 2025-06-13:
Hello guys

I'm dropping by to ask if there's any way I can contribute to the development or help with this project. I'm relatively new to kernel development (I have some experience with patching KVM for my own needs) and network devices, but I have a foundational understanding and am eager to learn more. With some guidance, I believe I could get up to speed and start making meaningful contributions.

I am maintaining a fork of `[rtl8852au](https://github.com/natimerry/rtl8852au)` to maintain compatibility with the newer kernel versions. This has sparked my interest in getting involved with a more structured active project where I can deepen my experience. 

If there are areas where a newcomer like me could help, I’d really appreciate any pointers or direction.

dubhater, 2025-06-13:
Yeah, jump right in! Create rtw8852au.c and try it out, fix the problems you encounter until it works.

You will run into the same problems fixed by these commits, plus a few others:

https://github.com/morrownr/rtw89/commit/f324143959a5d241ee3a2f6de4b7cdda54a3c700
https://github.com/morrownr/rtw89/commit/98489070a054bae606bf0cfea60f2b71d406e2fc
https://github.com/morrownr/rtw89/commit/887b18ba64658ec8c7aca7d1e2c51ce37bac9902
https://github.com/morrownr/rtw89/commit/b37c529144a58781636bb25e8d148cc779b1793c

natimerry, 2025-06-13:
Alright thanks for the starter tips, one more newbie question. I see a lot of magic numbers being thrown around, any tips on how I can research and find these stuff?

morrownr, 2025-06-13:
@natimerry 

Can you give us an example of a magic number?

natimerry, 2025-06-14:
> [@natimerry](https://github.com/natimerry)
> 
> Can you give us an example of a magic number?

Magic number probably isnt the write word but some stuff like this

https://github.com/morrownr/rtw89/blob/main/rtw8852a.c#L172

Where can I expect to find data like this for my chipset?

Kiwimannen, 2025-06-14:
I can send a RTL8912AU based adapter within Europe - to dubhater ? as I would love to see linux support.

You would have a choice between a Fenvi WiFi 7 BE6500 or a COMFAST CF-983BE - both work with MLO i Windows 11 using Asus ROG USB-BE92 drivers connected to a Asus BT8 router.

I have a spare RTL8832CU adapter too - but I can see that you already have one of those.

I have no idea how/if  you can send PM:s in Github but you can send me a mail to kiwimannen at yahoo dot se.

natimerry, 2025-06-14:
@dubhater I need some guidance. I have created [rtw8852au](https://github.com/natimerry/rtw89/commit/2ab3345f6f4a36aaf4ea8e5a77c4ab47789aef41) however, powering on the device seems to fail and I am unsure how I can proceed fixing that.

dubhater, 2025-06-14:
@Kiwimannen That would be great, thank you! I will email you, I don't think Github has private messages.

@natimerry

> powering on the device seems to fail and I am unsure how I can proceed fixing that.

Indeed, it doesn't power on. rtw89 has the function rtw89_mac_power_switch() in mac.c. You can start looking for the problem there.

> Im looking at the power on script for Larry's driver in platform/platform_RTK_DMP_usb.c on the rtl8852au tree

I don't know what that is, but you don't need it. You don't need anything from the platform folder. Everything you need is in the phl, os_dep, and include folders.

> https://github.com/morrownr/rtw89/blob/main/rtw8852a.c#L172
>
> Where can I expect to find data like this for my chipset?

grep can help with that. For example, you can find those tables in the other driver by looking for PWR_INTF_MSK_ALL.

I use this command a lot: ``grep --color=always -r -i -I -n 'thing' .|less -R`` If you have a proper IDE maybe you can use its built-in search feature instead of grep.

F-Poole, 2025-06-29:
Here, with a RTL8851BU USB dongle the conection lasts a few minutes and then I need to reload the driver to be able to connect again. I get this on dmesg:
> [ 5287.925575] rtw89_8851bu_git 1-9:1.2: rtw89_hw_scan_offload failed ret -110
[ 5289.075637] rtw89_8851bu_git 1-9:1.2: c2h reg timeout
[ 5289.094678] rtw89_8851bu_git 1-9:1.2: FW does not process h2c registers
[ 5289.309822] rtw89_8851bu_git 1-9:1.2: HW scan failed: -110
[ 5289.653496] rtw89_8851bu_git 1-9:1.2: Update probe request failed

a5a5aa555oo, 2025-07-01:
> The firmware folder here does not include all rtw89 firmware files. Can you elaborate on the reason for that?

I'm not good at English, I believe @dubhater could give you a good answer.

> I assume that I should replace the rtw8852c_fw-2.bin file with the new one?
> I do not know if the new rtw8922a_fw-4.bin needs to be added here. Can you clarify?

A little tip for you: Don't update the firmware files without updating the code. Our rtw89 driver here may be unable to work with the latest firmware.

Actually it's recommended to update the driver/firmware about every two months. After merging the patches from upstream, you have to do the compilation tests which may take you much time to ensure these new changes won't cause any compilation issue, new changes may also introduce new problems, so I prefer not to merge the upstream patches too frequently.

> If you would be so kind as to email me at morrownr at gmail dot com , I would appreciate it as I would like to discuss exactly how you take the patches and apply them to this repo.

Please see this [reply](https://github.com/morrownr/rtw89/issues/4#issuecomment-2894303048), did you run into any problems while merging the patches?

If you are not familiar with this, I will merge them when Ping-Ke send next PR to wireless-next (about 2~3 weeks later).

dubhater, 2025-07-01:
@F-Poole That should be fixed now.

dubhater, 2025-07-03:
RTL8852CU can be tested now.

When I tried it two months ago it couldn't go one day without losing the connection (``wlp3s0f3u4: Connection to AP 12:34:56:78:9a:bc lost``). I'm curious if it's better now with the newer firmware and driver.

morrownr, 2025-07-04:
@dubhater 

> I'm curious if it's better now with the newer firmware and driver.

Which newer firmware should we be testing? The one that came in 3 days ago:

rtw89/rtw8852c_fw-2.bin | Bin 2444568 -> 2461176 bytes

Nothing has been recently updated here in the repo.

> RTL8852CU can be tested now.

I am testing. Just need to know what firmware to use.

Also, I still have not figured out how this rtw89 firmware works.

a5a5aa555oo, 2025-07-04:
> There are some things I do not fully understand at this point.

@morrownr 

This is how I merge the patches from upstream.
https://drive.google.com/file/d/1svttchXOcGeHg5J_GDs2lzfnX5T-Job1/view?usp=sharing

@dubhater 

I'm curious how you merge them, is there a smarter way? `git am xx.patch` doesn't seem to work.

dubhater, 2025-07-04:
@morrownr I'm testing the one provided by Arch Linux when I last updated everything:

```
[   70.313066] rtw89_8852cu 1-4:1.0: loaded firmware rtw89/rtw8852c_fw-2.bin
[   70.354811] rtw89_8852cu 1-4:1.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 1
[   70.354823] rtw89_8852cu 1-4:1.0: Firmware version 0.27.125.0 (526f84c0), cmd version 0, type 3
```

But whatever firmware you have on your computer is fine. Any testing is useful.

@a5a5aa555oo I used to just copy the .c and .h files and then use Kate's git tools to stage and commit the relevant changes. The results can be messy, as you may have seen.

Recently I started using another way as well:

* Save the desired patches with QGit (saving multiple patches at a time may not be enabled by default, you have to check a box in the settings)
* Fix the paths in them: ``sed -i 's#drivers/net/wireless/realtek/rtw89/##g' /tmp/000*``
* Apply the modified patches: ``git am /tmp/000*``

dubhater, 2025-07-04:
Okay, it's not fixed. Moving on to firmware 0.27.129.1.

morrownr, 2025-07-04:
@dubhater 

> Okay, it's not fixed. Moving on to firmware 0.27.129.1.

I have not seen a dropped connection, I assume that is still the problem. It has been over 24 hours here and has survived multiple suspend cycles. I am on Ubuntu 25.04 with kernel 6.14.

What firmware am I using? Firmware version 0.27.125.0

Is firmware 0.27.129.1 the one that came in about 4 days ago?

dubhater, 2025-07-04:
It loses the connection. This time it took 5.5 hours of normal network activity. You may not notice anything happened because it reconnects right away.

I believe you can make it happen faster using iperf3:

```sh
iperf3 -c 192.168.0.1 -R -t 3000
```

Yes, 0.27.129.1 is the one from 4 days ago.

morrownr, 2025-07-04:
@dubhater 

> You may not notice anything happened because it reconnects right away.

Ah. Okay. Will pay attention to the log.

> Yes, 0.27.129.1 is the one from 4 days ago.

I'll move to 0.27.129.1 and continue testing. Which band are you seeing the problem on?

dubhater, 2025-07-04:
2.4 and 5 GHz both. (I don't have a 6 GHz AP.)

morrownr, 2025-07-04:
@dubhater 

> I don't have a 6 GHz AP.

You have a couple of adapters with mt7921au chips so you have a 6 GHz AP. ;)

I have problems with that solution in my country due to the regdb entries for my country but in the EU, you should not have that problem.

Questions:

Why am I seeing two lines?

```
[    4.330022] rtw89_8852cu_git 5-2.3:1.0: Firmware version 0.27.129.1 (a71be205), cmd version 0, type 1
[    4.330028] rtw89_8852cu_git 5-2.3:1.0: Firmware version 0.27.129.1 (a71be205), cmd version 0, type 3

```

Never seen a line like that. What's going on?

[    4.716788] rtw89_8852cu_git 5-2.3:1.0: rfkill hardware state changed to enable

a5a5aa555oo, 2025-07-05:
> Recently I started using another way as well:
> ...

@dubhater 

This method is a little complicated to me but looks much smarter!

[44 more replies]