morrownr/rtw89 #7: "options rtw89_core_git disable_ps_mode=y" required otherwise connection lost regularly

BennyE, 2025-05-25:
Hi,

I wanted to report that I made the Realtek 8922AE work with Multi-Link Operation (MLO) on my Raspberry Pi 5 with 52Pi E-Key PCIe Top/HAT.

I noticed that the connection to my AP would be unstable, unless I set "options rtw89_core_git disable_ps_mode=y".
Kernel: 6.14.7-v8-16k+

1. What do you need from me to identify/resolve the disable_ps_mode=n/y issue?
2. Why does the driver decide to associate to 5 GHz if 6 GHz is available?
3. Under which circumstances would the NIC use STR? (I only saw packets in the radio/band the client associated to)
4. Why does the output of "iw" lack the "rx bitrate", while it is there with 6.14 vanilla kernel?
pi@wifitux:~/may25/rtw89 $ cat /etc/modprobe.d/rtw89.conf 
```
options rtw89_core_git debug_mask=0x0
#options rtw89_core_git disable_ps_mode=n
options rtw89_core_git disable_ps_mode=y

options rtw89_pci_git disable_clkreq=n
options rtw89_pci_git disable_aspm_l1=n
options rtw89_pci_git disable_aspm_l1ss=n

# Blacklist the in-kernel rtw89 drivers
blacklist rtw89_8851be
blacklist rtw89_8851b
blacklist rtw89_8852ae
blacklist rtw89_8852a
blacklist rtw89_8852b_common
blacklist rtw89_8852be
blacklist rtw89_8852b
blacklist rtw89_8852bte
blacklist rtw89_8852bt
blacklist rtw89_8852ce
blacklist rtw89_8852c
blacklist rtw89_8922ae
blacklist rtw89_8922a
blacklist rtw89_core
blacklist rtw89_pci
```

lspci -vvv
```
0001:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8922AE 802.11be PCIe Wireless Network Adapter (rev 01)
	Subsystem: Realtek Semiconductor Co., Ltd. RTL8922AE 802.11be PCIe Wireless Network Adapter
	Control: I/O- Mem+ BusMaster+ SpecCycle- MemWINV- VGASnoop- ParErr- Stepping- SERR- FastB2B- DisINTx+
	Status: Cap+ 66MHz- UDF- FastB2B- ParErr- DEVSEL=fast >TAbort- <TAbort- <MAbort- >SERR- <PERR- INTx-
	Latency: 0
	Interrupt: pin A routed to IRQ 180
	Region 2: Memory at 1b80000000 (64-bit, non-prefetchable) [size=1M]
	Capabilities: [40] Power Management version 3
		Flags: PMEClk- DSI- D1- D2- AuxCurrent=375mA PME(D0+,D1-,D2-,D3hot+,D3cold+)
		Status: D0 NoSoftRst+ PME-Enable- DSel=0 DScale=0 PME-
	Capabilities: [50] MSI: Enable+ Count=1/8 Maskable+ 64bit+
		Address: 00000000fffffffc  Data: 6540
		Masking: 000000fe  Pending: 00000000
	Capabilities: [70] Express (v2) Endpoint, IntMsgNum 0
		DevCap:	MaxPayload 256 bytes, PhantFunc 0, Latency L0s <4us, L1 <64us
			ExtTag- AttnBtn- AttnInd- PwrInd- RBE+ FLReset+ SlotPowerLimit 0W TEE-IO-
		DevCtl:	CorrErr+ NonFatalErr+ FatalErr+ UnsupReq+
			RlxdOrd+ ExtTag- PhantFunc- AuxPwr- NoSnoop- FLReset-
			MaxPayload 256 bytes, MaxReadReq 512 bytes
		DevSta:	CorrErr- NonFatalErr- FatalErr- UnsupReq- AuxPwr+ TransPend-
		LnkCap:	Port #0, Speed 5GT/s, Width x1, ASPM L0s L1, Exit Latency L0s <2us, L1 <64us
			ClockPM+ Surprise- LLActRep- BwNot- ASPMOptComp+
		LnkCtl:	ASPM L0s L1 Enabled; RCB 64 bytes, LnkDisable- CommClk+
			ExtSynch- ClockPM+ AutWidDis- BWInt- AutBWInt-
		LnkSta:	Speed 5GT/s, Width x1
			TrErr- Train- SlotClk+ DLActive- BWMgmt- ABWMgmt-
		DevCap2: Completion Timeout: Not Supported, TimeoutDis+ NROPrPrP- LTR+
			 10BitTagComp- 10BitTagReq- OBFF Not Supported, ExtFmt- EETLPPrefix-
			 EmergencyPowerReduction Not Supported, EmergencyPowerReductionInit-
[... 190 lines cut ...]
```
support@AP-CE:40:~$ ssudo sta_list
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
96:b4:ea:71:2b:49  192.168.2.115                                               683                 0             0             2.4GHz  PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
3a:c0:57:fb:1b:8a  192.168.2.115                                               683                 0             0             6GHz    PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
f6:35:1f:ec:c1:df  192.168.2.115                                               683                 1375481659    10271782      5GHz    PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
support@AP-CE:40:~$ 
support@AP-CE:40:~$ 
support@AP-CE:40:~$ ssudo sta_list
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
96:b4:ea:71:2b:49  192.168.2.115                                               2646                0             0             2.4GHz  PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
3a:c0:57:fb:1b:8a  192.168.2.115                                               2646                0             0             6GHz    PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
SSID:Stellar-Wi-Fi-7-Pi
STA_MAC                  IPv4                       IPv6                       OnlineTime             RX            TX         FREQ      AUTH            Final_role            VLANID     TUNNELID      FARENDIP
f6:35:1f:ec:c1:df  192.168.2.115                                               2646                1375482006    10276366      5GHz    PSK_WPA3      __Stellar-Wi-Fi-7-Pi      0           0             
support@AP-CE:40:~$ 
```

morrownr, 2025-06-01:
@a5a5aa555oo 

I don't have any Realtek cards. Is this an issue that should go directly to linux-wireless and Ping Ke?

a5a5aa555oo, 2025-06-01:
@morrownr 

I think yes.

@BennyE 

Please file a bug report to linux-wireless ([linux-wireless@vger.kernel.org](mailto:linux-wireless@vger.kernel.org)) directly, only realtek developers are able to answer these questions.

Examples:
https://lore.kernel.org/linux-wireless/CAD+XiyEJo0m8L6PbmBD=NPPa0E_wX3kk4ezn1qZ0d8JGXsS1qQ@mail.gmail.com/
https://lore.kernel.org/linux-wireless/uidltsdsuujrczrtzgerhh5cw2tztxktfen6yvztnc7gttzgvk@jccomj7f4gul/
https://lore.kernel.org/linux-wireless/20240917055730.Horde.pVyI-XF1MRA_zUQ5BDJZ4lC@ignaz.org/

morrownr, 2025-06-01:
@BennyE 

> Please file a bug report to linux-wireless (linux-wireless@vger.kernel.org) directly...

Your email client will need to be in text mode, not html, or the server will reject the email.

morrownr, 2025-06-01:
@BennyE 

Also, use a subject line like the following:

[Bug Report] wifi: rtw89: 8922ae