morrownr/rtw89 #9: Can't connect to any Wifi, Always Show Configuring Interface

VisionaryAppDev, 2025-06-07:
Hi @morrownr,

I bought the Comfast CF-943AX recently and I couldn't get it working on my Arch Linux. Recently, I have installed the driver in this repo after I found some discussion from [USB-WiFi issues 510](https://github.com/morrownr/USB-WiFi/issues/510). It does seem like now I could scan the Wifi but when I tired to connect, it just show Configuring Interface and after like 30s, it asks for a password and it just repeat this again and again until I cancel. At the same time, for bluetooth, it doesn't work like other people has report. It only connected, play sound for 5 second and disconnected. If I tried again, it just repeated, disconnected after 5 seconds.

By the way, previously, I have tried installing the one from [fofajardo/rtl8851bu/issues/5](https://github.com/fofajardo/rtl8851bu/issues/5) and [heesn/rtl8831/issues/11](https://github.com/heesn/rtl8831/issues/11) and it get the same issue (Configuring Interface), but bluetooth is working just fine when I use these two driver.

Now, let come back to detail configure I got:

lsusb:
```
Bus 001 Device 005: ID 0bda:b851 Realtek Semiconductor Corp. 802.11ax WLAN Adapter
```

dkms status:
```
rtw89/6.16, 6.12.31-1-lts, x86_64: installed
```

ip a:
```
...
3: wlp14s0u3i2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
..
altname wlx40a5ef2fbb68
```

iwconfig:
```
wlp14s0u3i2  IEEE 802.11  ESSID:off/any
          Mode:Managed  Access Point: Not-Associated   Tx-Power=off
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:on
```

sudo lshw -C network: 
```
  *-network
       description: Wireless interface
       product: 802.11ax WLAN Adapter
       physical id: 5
       bus info: usb@1:3
       logical name: wlp14s0u3i2
       serial: 40:xx:68
       capabilities: ethernet physical wireless
       configuration: broadcast=yes driver=rtw89_8851bu_git driverversion=6.12.31-1-lts firmware=N/A link=no multicast=yes wireless=IEEE 802.11
```

cat /etc/NetworkManager/NetworkManager.conf
```
# Configuration file for NetworkManager.
# See "man 5 NetworkManager.conf" for details.

[device]
wifi.scan-rand-mac-address=no

```

When I tried to connect to any wifi manually using wpa_supplicant command `sudo wpa_supplicant -i wlp14s0u3i2 -c wpa_supplicant.example -d`:
```
## wpa_supplicant.example Config:
ctrl_interface=/run/wpa_supplicant
ctrl_interface_group=wheel
update_config=1
country=World
ap_scan=1

network={
ssid="1"
psk="raw-password"
}
[... 217 lines cut ...]
wlp14s0u3i2: State: DISCONNECTED -> DISCONNECTED
nl80211: Set wlp14s0u3i2 operstate 0->0 (DORMANT)
netlink: Operstate: ifindex=3 linkmode=-1 (no change), operstate=5 (IF_OPER_DORMANT)
EAPOL: External notification - portEnabled=0
EAPOL: External notification - portValid=0
QM: Clear all active DSCP policies
wlp14s0u3i2: CTRL-EVENT-DSCP-POLICY clear_all
wlp14s0u3i2: WPA: Clear old PMK and PTK
wlp14s0u3i2: Cancelling scan request
wlp14s0u3i2: Cancelling authentication timeout
Off-channel: Clear pending Action frame TX (pending_action_tx=(nil)
nl80211: Data frame filter flags=0x0
Off-channel: Action frame sequence done notification: pending_action_tx=(nil) drv_offchan_tx=1 action_tx_wait_time=0 off_channel_freq=0 roc_waiting_drv_freq=0
QM: Clear all active DSCP policies
wlp14s0u3i2: CTRL-EVENT-DSCP-POLICY clear_all
Remove interface wlp14s0u3i2 from radio phy0
Remove radio phy0
nl80211: deinit ifname=wlp14s0u3i2 disabled_11b_rates=0
nl80211: Remove monitor interface: refcount=0
netlink: Operstate: ifindex=3 linkmode=0 (kernel-control), operstate=6 (IF_OPER_UP)
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Unsubscribe mgmt frames handle 0x8888ec1a629b3e99 (mode change)
```

So, do you know how may I get it fix?

VisionaryAppDev, 2025-06-07:
Now, I have disabled wpa_supplicant and NetworkManager service and try again using wpa_supplicant cli and got this logs:
```
wpa_supplicant v2.11-hostap_2_11+
Successfully initialized wpa_supplicant
Initializing interface 'wlp14s0u3i2' conf 'wpa_supplicant.example' driver 'default' ctrl_interface 'N/A' bridge 'N/A'
Configuration file 'wpa_supplicant.example' -> '/home/yourpc/wpa_supplicant.example'
Reading configuration file '/home/yourpc/wpa_supplicant.example'
ctrl_interface='/run/wpa_supplicant'
ctrl_interface_group='wheel'
update_config=1
country='Wo'
ap_scan=1
Priority group 0
   id=0 ssid='2'
nl80211: Kernel version: Linux 6.12.31-1-lts (#1 SMP PREEMPT_DYNAMIC Thu, 29 May 2025 09:39:59 +0000; x86_64)
nl80211: Maximum supported attribute ID: 332
nl80211: Initialize interface wlp14s0u3i2 (driver: rtw89_8851bu_git)
nl80211: TDLS supported
nl80211: TDLS external setup
nl80211: Supported cipher 00-0f-ac:1
nl80211: Supported cipher 00-0f-ac:5
nl80211: Supported cipher 00-0f-ac:2
nl80211: Supported cipher 00-0f-ac:4
nl80211: Supported cipher 00-0f-ac:10
nl80211: Supported cipher 00-0f-ac:8
nl80211: Supported cipher 00-0f-ac:9
nl80211: Supported cipher 00-0f-ac:6
nl80211: Supported cipher 00-0f-ac:13
nl80211: Supported cipher 00-0f-ac:11
nl80211: Supported cipher 00-0f-ac:12
nl80211: Using driver-based off-channel TX
nl80211: Driver-advertised extended capabilities (default) - hexdump(len=8): 04 00 40 00 00 00 00 40
nl80211: Driver-advertised extended capabilities mask (default) - hexdump(len=8): 04 00 40 00 00 00 00 40
nl80211: Use separate P2P group interface (driver advertised support)
nl80211: key_mgmt=0x1ff0f enc=0xfef auth=0x7 flags=0x24000510535bfae0 flags2=0x103 rrm_flags=0x30 probe_resp_offloads=0x0 max_stations=0 max_remain_on_chan=5000 max_scan_ssids=8
nl80211: interface wlp14s0u3i2 in phy phy0
nl80211: Set mode ifindex 3 iftype 2 (STATION)
nl80211: Subscribe to mgmt frames with non-AP handle 0x621c678a1610
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0104 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=040a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=040b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=040c multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=040d multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=090a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=090b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=090c multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=090d multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0409506f9a09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=7f506f9a09 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0409506f9a1a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0800 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0801 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=040e multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=12 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=06 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0a07 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0a11 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0a0b multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0a1a multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=1101 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=1102 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0505 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=0500 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=1301 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=1305 multicast=0
nl80211: Register frame type=0xd0 (WLAN_FC_STYPE_ACTION) nl_handle=0x621c678a1610 match=7e506f9a1a multicast=0
rfkill: initial event: idx=1 type=1 op=0 soft=0 hard=0
netlink: Operstate: ifindex=3 linkmode=1 (userspace-control), operstate=5 (IF_OPER_DORMANT)
Add interface wlp14s0u3i2 to a new radio phy0
nl80211: Regulatory information - country=00
[... 235 lines cut ...]
wlp14s0u3i2: RSN: clearing own RSNXE
nl80211: Associate (ifindex=3)
  * bssid=6c:xxxxx:a3
  * freq=2467
  * SSID=2
  * IEs - hexdump(len=66): 30 14 01 00 00 0f ac 04 01 00 00 0f ac 04 01 00 00 0f ac 02 0c 00 7f 0a 04 00 4a 02 01 40 00 40 00 01 46 05 70 00 00 00 00 3b 17 51 51 52 53 54 73 74 75 76 77 78 79 7a 7b 7c 7d 7e 7f [... cut ...]
wlp14s0u3i2: SME: Association request to the driver failed
wlp14s0u3i2: Radio work 'sme-connect'@0x621c67928c90 done in 35.075165 seconds
wlp14s0u3i2: radio_work_free('sme-connect'@0x621c67928c90): num_active_works --> 0
wlp14s0u3i2: Added BSSID 6c:xxxxx:a3 into ignore list, ignoring for 10 seconds
wlp14s0u3i2: Consecutive connection failures: 1 --> request scan in 100 ms
wlp14s0u3i2: Setting scan request: 0.100000 sec
wlp14s0u3i2: State: ASSOCIATING -> DISCONNECTED
nl80211: Set wlp14s0u3i2 operstate 0->0 (DORMANT)
```

morrownr, 2025-06-09:
Hi @VisionaryAppDev 

I'd like to start off simple. I need to know what chip is in your adapter. Please run and post the results of the following:

$ lsusb

What is it that you are trying to do? Managed mode?

You mentioned that you had previously installed 2 other drivers. Did you totally uninstall them? This can be a source of problems.

VisionaryAppDev, 2025-06-12:
Thank @morrownr  and  @dubhater,

Now, it's working but I couldn't identify which step fix it though. First, I have updated my system as usual and this time it's also include Kernel and kernel-header. After the reboot, I check `sudo lshw -C network` and found a driver from Comfast (https://en.comfast.com.cn/uploadfile/driver/RTL8851BU_WiFi_linux_v1.19.10.zip) that I previously install (although it's failed to compile during `sudo make install` due to my Kernel or C programming version is newer and deprecate sth I don't understand). Then, I tried to uninstall it both from `dkms`, `/usr/src/rtl8851..`, `/lib/modules/6.12.31-1-lts/updates/dkms/88` and other possible option that ChatGPT give. At the same time, I switch from wpa_supplicant to iwd because I doubt if it's because of wpa_suppliant. And then stop wpa_supplicant, restart NetworkManager and nothing happen. But, after reboot, it is still doesn't work. It's just showing connecting icon as usual. Unexpectedly, after like 5mn, it's show connected and wow, I feel like magic is happening. Now, I tried switching back to wpa_supplicant it confirm it's working properly.

But, there is some problem I still got, upon restart, it doesn't seem like the Wifi adapter is activated immediately. On my network icon, it shown nothing and like 3mn before the Wifi icon appears and take like 2 or 3mn (if i click on NetworkManager Icon to manually scan and connect to my Wifi, my GUI stuck again and again) more before it could connect to saved Wifi. And when I click on the Network Icon (network manager), mostly it stuck the whole PC from browser to Terminal and most of other thing. I couldn't do anything, even restart the PC won't work. It just showing count down for a pretty long time, until I force shutdown by pressing the button. Not sure if @dubhater also got this issue as well or not?

morrownr, 2025-06-12:
@VisionaryAppDev 

> Now, it's working but I couldn't identify which step fix it though. First, I have updated my system as usual and this time it's also include Kernel and kernel-header.

It is likely that the new kernel does not have a previously installed driver that is interfering with this rtw89 driver. The Realtek out-of-kernel (vendor) drivers are simply not suitable for everyday computer users. They cause problems and we need to get away from them. That is what this repo is all about. Something that might help is if you read some items at the Main Menu of this site:

https://github.com/morrownr/USB-WiFi

Menu item 1 contains good information. It is background information that is good for Linux users to know. Menu item 2 is helpful to users in that it contains information about many adapters that are easy to use.

Now back to the topic:

> and other possible option that ChatGPT give.

My experience is that AI does not do a very good job with Linux wireless. AI is simply using information that is available and there are problems with the information that is available. The problems center around several things like many changes over time, old information is often not good due to the many changes to Linux wireless over the last several years. There is simply so much bad information on the internet that AI has problems sorting the good information from the bad. Use caution when using AI with Linux wireless.

> But, there is some problem I still got, upon restart, it doesn't seem like the Wifi adapter is activated immediately. 

What distro are you using?

What kernel does your distro use?

$ uname -r

Run and post the results of the following command:

$ sudo dmesg | grep 8851

VisionaryAppDev, 2025-06-13:
@morrownr, here is the result: 

>  What kernel does your distro use?

I use Arch Linux, and here is my current kernel version "6.12.33-1-lts".

> sudo dmesg | grep 8851

```
[   49.291589] Bluetooth: hci0: RTL: examining hci_ver=0c hci_rev=000b lmp_ver=0c lmp_subver=8851
[   49.294594] Bluetooth: hci0: RTL: loading rtl_bt/rtl8851bu_fw.bin
[   49.294954] Bluetooth: hci0: RTL: loading rtl_bt/rtl8851bu_config.bin
[   49.310786] rtw89_8851bu_git 1-3:1.2: loaded firmware rtw89/rtw8851b_fw.bin
[   49.436514] rtw89_8851bu_git 1-3:1.2: MAC has already powered on
[   50.804165] rtw89_8851bu_git 1-3:1.2: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 5
[   50.804172] rtw89_8851bu_git 1-3:1.2: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 3
[   58.765391] rtw89_8851bu_git 1-3:1.2: chip rfe_type is 4
[   59.758395] rtw89_8851bu_git 1-3:1.2: rfkill hardware state changed to enable
[   59.758460] usbcore: registered new interface driver rtw89_8851bu_git
```

VisionaryAppDev, 2025-06-14:
I have looked through the BIOS on my ASRock B650M PG Riptide and here is what I can find
- xHCI Hand-off: which is already enabled
- USB Mass Storage Driver:  already already enable as well

I have switch from USB 3 to USB 2 and it doesn't seem to make any different though. But, what I notice when it's stuck (when click on Wifi Icon), it seem like something wrong with the USB port. During the stuck, if I unplug the Wifi Adapter and switch the port between keyboard (USB 3.X), my keyboard show no light and not working at all. Event if I plug the keyboard to the original port, it has no light and when I press any key, nothing work.

a5a5aa555oo, 2025-06-14:
@VisionaryAppDev 

I saw some problems in your recording.

1. at 0:48 you removed /etc/modprobe.d/rtw89.conf. This is not needed if you want to update the rtw89 driver. `/etc/modprobe.d/rtw89.conf` is necessary for the driver, do you copy it back to /etc/modprobe.d/ after removing it? 

2. at 1:25 after updating the driver, the new changes won't take effect right away, you have to reboot or reload the driver manually.

This command can unload the whole driver for 8851BU
```
sudo rmmod rtw89_{8851bu,8851b,usb,core}_git
```

This command can load the driver for 8851BU
```
sudo modprobe rtw89_8851bu_git
```

Could you post the output of the command `lsmod | grep rtw89`?  I'm curious if the correct modules (drivers) were loaded in your system.

dubhater, 2025-06-14:
I bet dmesg shows the same "rtw89_hw_scan_offload failed ret -110" error others have reported.

VisionaryAppDev, 2025-06-15:
@dubhater,

I have just got the time and review journal ctl log and found the following trace:
```
Jun 14 20:59:48 yourpc sudo[3985]:   yourpc : TTY=pts/1 ; PWD=/home/yourpc/rtw89 ; USER=root ; COMMAND=/usr/bin/rm -rf /usr/src/rtw89-6.16/
Jun 14 20:59:48 yourpc sudo[3985]: pam_unix(sudo:session): session opened for user root(uid=0) by yourpc(uid=1000)
Jun 14 20:59:48 yourpc sudo[3985]: pam_unix(sudo:session): session closed for user root
Jun 14 21:00:00 yourpc sudo[3995]:   yourpc : TTY=pts/1 ; PWD=/home/yourpc/rtw89 ; USER=root ; COMMAND=/usr/bin/rm -f /etc/modprobe.d/rtw89.conf
Jun 14 21:00:00 yourpc sudo[3995]: pam_unix(sudo:session): session opened for user root(uid=0) by yourpc(uid=1000)
Jun 14 21:00:00 yourpc sudo[3995]: pam_unix(sudo:session): session closed for user root
Jun 14 21:00:00 yourpc iwd[2244]: event: roam-scan,
Jun 14 21:00:04 yourpc iwd[2244]: event: roam-scan,
Jun 14 21:00:15 yourpc iwd[2244]: event: roam-scan,
Jun 14 21:00:18 yourpc iwd[2244]: event: roam-scan,
Jun 14 21:00:19 yourpc sudo[4002]:   yourpc : TTY=pts/1 ; PWD=/home/yourpc/rtw89 ; USER=root ; COMMAND=/usr/bin/dkms install /home/yourpc/rtw89
Jun 14 21:00:19 yourpc sudo[4002]: pam_unix(sudo:session): session opened for user root(uid=0) by yourpc(uid=1000)
Jun 14 21:00:32 yourpc sudo[4002]: pam_unix(sudo:session): session closed for user root
Jun 14 21:00:46 yourpc NetworkManager[693]: <info>  [1749909646.2567] audit: op="statistics" interface="wlan0" ifindex=4 args="2000" pid=1135 uid=1000 result="success"
Jun 14 21:00:46 yourpc NetworkManager[693]: <info>  [1749909646.2585] audit: op="statistics" interface="wlan0" ifindex=4 args="500" pid=1205 uid=1000 result="success"
Jun 14 21:00:55 yourpc NetworkManager[693]: <info>  [1749909655.7768] device (wlan0): state change: activated -> unavailable (reason 'supplicant-failed', managed-type: 'full')
Jun 14 21:00:55 yourpc NetworkManager[693]: <info>  [1749909655.7768] dhcp4 (wlan0): canceled DHCP transaction
Jun 14 21:00:55 yourpc NetworkManager[693]: <info>  [1749909655.7768] dhcp4 (wlan0): activation: beginning transaction (timeout in 45 seconds)
Jun 14 21:00:55 yourpc NetworkManager[693]: <info>  [1749909655.7768] dhcp4 (wlan0): state changed no lease
Jun 14 21:00:55 yourpc systemd[1]: iwd.service: Main process exited, code=killed, status=11/SEGV
Jun 14 21:00:55 yourpc systemd[1]: iwd.service: Failed with result 'signal'.
Jun 14 21:00:55 yourpc kernel: rtw89_8851bu_git 1-12:1.2: avoiding multiple of 512
Jun 14 21:00:55 yourpc kernel: skbuff: skb_over_panic: text:ffffffffc1d289ff len:492 put:4 head:ffff8bf2f8fe5a80 data:ffff8bf2f8fe5ad8 tail:0x244 end:0x240 dev:<NULL>
Jun 14 21:00:55 yourpc kernel: ------------[ cut here ]------------
Jun 14 21:00:55 yourpc kernel: kernel BUG at net/core/skbuff.c:209!
Jun 14 21:00:55 yourpc kernel: Oops: invalid opcode: 0000 [#1] PREEMPT SMP NOPTI
Jun 14 21:00:55 yourpc kernel: CPU: 9 UID: 0 PID: 2244 Comm: iwd Tainted: G           OE      6.12.33-1-lts #1 5a18395dfdf04747926284a14c4457e4c6f0a51a
Jun 14 21:00:55 yourpc kernel: Tainted: [O]=OOT_MODULE, [E]=UNSIGNED_MODULE
Jun 14 21:00:55 yourpc kernel: Hardware name: ASRock B650M PG Riptide/B650M PG Riptide, BIOS 3.20 02/21/2025
Jun 14 21:00:55 yourpc kernel: RIP: 0010:skb_panic+0x55/0x57
Jun 14 21:00:55 yourpc kernel: Code: 89 d2 50 8b 87 c0 00 00 00 50 8b 87 bc 00 00 00 50 ff b7 d0 00 00 00 4c 8b 8f c8 00 00 00 48 c7 c7 60 f5 d8 8d e8 db e3 f9 ff <0f> 0b 48 8b 14 24 48 c7 c1 10 55 81 8d e8 99 ff ff ff 48 8b 14 24
Jun 14 21:00:55 yourpc kernel: RSP: 0018:ffffd0b745d477d0 EFLAGS: 00010246
Jun 14 21:00:55 yourpc kernel: RAX: 0000000000000087 RBX: ffffd0b745d47848 RCX: 0000000000000027
Jun 14 21:00:55 yourpc kernel: RDX: 0000000000000000 RSI: 0000000000000001 RDI: ffff8bf95e6a1880
Jun 14 21:00:55 yourpc kernel: RBP: ffffd0b745d47848 R08: 0000000000000000 R09: 00000000ffffefff
Jun 14 21:00:55 yourpc kernel: R10: ffffffff8e65d540 R11: ffffd0b745d47670 R12: 0000000000000018
Jun 14 21:00:55 yourpc kernel: R13: ffff8bf2cd9a2000 R14: ffff8bf2cd9a2408 R15: ffffd0b745d47878
Jun 14 21:00:55 yourpc kernel: FS:  00007cb5e14a8b80(0000) GS:ffff8bf95e680000(0000) knlGS:0000000000000000
Jun 14 21:00:55 yourpc kernel: CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Jun 14 21:00:55 yourpc kernel: CR2: 00007fa30e397000 CR3: 000000012a4f0000 CR4: 0000000000f50ef0
Jun 14 21:00:55 yourpc kernel: PKRU: 55555554
Jun 14 21:00:55 yourpc kernel: Call Trace:
Jun 14 21:00:55 yourpc kernel:  <TASK>
Jun 14 21:00:55 yourpc kernel:  skb_put.cold+0x10/0x10
Jun 14 21:00:55 yourpc kernel:  rtw89_usb_ops_tx_write.cold+0x27/0xa8 [rtw89_usb_git e5a7d6cb805b8b64c14422dc5c0f0e7b95f0b54a]
Jun 14 21:00:55 yourpc kernel:  rtw89_h2c_tx+0xa0/0x170 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
Jun 14 21:00:55 yourpc kernel:  ? srso_alias_return_thunk+0x5/0xfbef5
Jun 14 21:00:55 yourpc kernel:  rtw89_h2c_tx_and_wait+0x1e/0x60 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
Jun 14 21:00:55 yourpc kernel:  rtw89_fw_h2c_scan_list_offload_ax+0x1b9/0x210 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
Jun 14 21:00:55 yourpc kernel:  ? rtw89_chanctx_pause+0x38/0x70 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
Jun 14 21:00:55 yourpc kernel:  rtw89_hw_scan_add_chan_list_ax+0xdd/0x190 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
Jun 14 21:00:55 yourpc kernel:  rtw89_hw_scan_offload+0x91/0x130 [rtw89_core_git 1a21b4fb50d1cf46ba1bcf8804895d58f2c6b046]
[... cut ...]
Jun 14 21:02:00 yourpc systemd[840]: Started Dolphin - File Manager.
Jun 14 21:02:01 yourpc kioworker[6019]: kf.kio.core.connection: Socket not connected QLocalSocket::PeerClosedError
Jun 14 21:02:01 yourpc kioworker[6019]: kf.kio.core: An error occurred during write. The worker terminates now.
Jun 14 21:02:02 yourpc kioworker[6063]: kf.kio.core.connection: Socket not connected QLocalSocket::PeerClosedError
Jun 14 21:02:02 yourpc kioworker[6063]: kf.kio.core: An error occurred during write. The worker terminates now.

```

dubhater, 2025-06-15:
"skbuff: skb_over_panic:" <- That could explain why things are freezing. I think it's fixed now.

VisionaryAppDev, 2025-06-17:
@dubhater,

Now, clicking on NetworkManager icon is no longer stuck. But, when boot up, it's stuck for like 20 second. During this period, I just saw blank screen, and none responsive when I press any keyboard shortcut.

Other than that, connecting and switching Wifi connection take a bit long. Please video the video below. And lastly, I experience some disconnecting without any reason. I saw some log but don't know if it's related. I will share more if I notice anything.
```
[  324.151980] rtw89_8851bu_git 1-3:1.2: avoiding multiple of 512
[ 1026.933113] Bluetooth: Unexpected continuation frame (len 0)
[ 1149.650999] iwd[837]: segfault at a21dd09f2 ip 00006120b16467aa sp 00007ffd5d8302b0 error 4 in iwd[2d7aa,6120b1622000+a5000] likely on CPU 9 (core 3, socket 0)
```

https://github.com/user-attachments/assets/61fa046d-c7ce-4d56-8e35-27f275307752

VisionaryAppDev, 2025-07-03:
But, seem like connecting to the WIFI still take a while. Switching also take a while too. Please check the log and video below:

https://github.com/user-attachments/assets/cd794e1d-49ad-4515-a065-f59e4f0fb4e9

[dmesg.txt](https://github.com/user-attachments/files/21039535/dmesg.txt)

dubhater, 2025-07-03:
Yes, something is still wrong.

It's worth trying these things again, now that some problems in rtw89 are fixed:

> Is the wireless-regdb package installed? Is your country code set correctly? Check with iw reg get, modify it with iw reg set XX. Maybe remove country=World from your wpa_supplicant configuration.
>
> If nothing else helps, move your network to channel 1, 6, or 11, whichever is less occupied. I think it will work if you don't use channels 12 and 13.

Also, if you're using WPA3, try WPA2 instead.

VisionaryAppDev, 2025-07-16:
> Do you have another wifi adapter that you can try, with a chip handled by a different driver?

I do have another older wifi adapter but let me find and try it if it is supported and work.

> Try to disable the encryption on the AP.

Is that you mean, disabled security, correct? Like without wpa or wpa2?

dubhater, 2025-07-16:
Yes, that's right.

VisionaryAppDev, 2025-07-30:
Ok, I have tried create another AP using different channel and now it's working although it takes quite a while. And here is the log from `iw event -T`

```
[2025-07-30 21:24:26.785499]: wlan0 (phy #0): scan finished: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-07-30 21:24:27.363816]: wlan0 (phy #0): scan started
[2025-07-30 21:24:30.298735]: wlan0: del station 80:xxx:f2
[2025-07-30 21:24:33.521347]: wlan0 (phy #0): deauth 40:xxxx:68 -> 80:xxx:f2 reason 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-07-30 21:24:33.521386]: wlan0 (phy #0): disconnected (local request) reason: 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-07-30 21:24:33.598828]: wlan0 (phy #0): scan aborted: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472, "0"
[2025-07-30 21:24:33.598861]: phy #0: regulatory domain change: set to world roaming by the wireless core upon initialization request
[2025-07-30 21:24:57.871132]: wlan0 (phy #0): scan started
[2025-07-30 21:25:00.802176]: wlan0 (phy #0): scan finished: 2417 2422 2427 2432 2442 2447 2452 2457 2467 2472 2484, ""
[2025-07-30 21:25:25.126266]: wlan0 (phy #0): scan started
[2025-07-30 21:25:26.825494]: wlan0 (phy #0): scan finished: 2412 2437 2462, ""
[2025-07-30 21:25:51.085788]: wlan0 (phy #0): scan started
[2025-07-30 21:25:51.160672]: phy #0: beacon hint:
phy0 5180 MHz [36]:
        o Initiating radiation enabled
[2025-07-30 21:25:53.428823]: wlan0 (phy #0): scan finished: 5180, "0"
[2025-07-30 21:26:28.317393]: wlan0: new station 80:xxx:f2
[2025-07-30 21:26:29.166410]: wlan0 (phy #0): auth 80:xxx:f2 -> 40:xxxx:68 status: 0: Successful
[2025-07-30 21:26:29.235492]: wlan0: del station 80:xxx:f2
[2025-07-30 21:26:31.174474]: wlan0 (phy #0): deauth 40:xxxx:68 -> 80:xxx:f2 reason 3: Deauthenticated because sending station is leaving (or has left) the IBSS or ESS
[2025-07-30 21:26:31.174518]: wlan0 (phy #0): timed out to 80:xxx:f2
[2025-07-30 21:26:55.437039]: wlan0 (phy #0): scan started
[2025-07-30 21:27:00.328811]: wlan0 (phy #0): scan finished: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-07-30 21:27:35.218310]: wlan0: new station 80:xxx:f2
[2025-07-30 21:27:36.067309]: wlan0 (phy #0): auth 80:xxx:f2 -> 40:xxxx:68 status: 126: <unknown>
[2025-07-30 21:27:36.140153]: wlan0 (phy #0): scan started
[2025-07-30 21:27:37.522140]: wlan0 (phy #0): scan finished: 5180, "0"
[2025-07-30 21:27:37.524088]: wlan0 (phy #0): auth 80:xxx:f2 -> 40:xxxx:68 status: 0: Successful
[2025-07-30 21:27:38.009373]: wlan0 (phy #0): assoc 80:xxx:f2 -> 40:xxxx:68 status: 0: Successful
[2025-07-30 21:27:38.009412]: wlan0 (phy #0): connected to 80:xxx:f2
[2025-07-30 21:27:38.009426]: wlan0 (phy #0): CQM event: RSSI (-80 dBm) went below threshold
[2025-07-30 21:27:38.090362]: phy #0: regulatory domain change: set to KR by a country IE request on phy0
[2025-07-30 21:27:38.090630]: wlan0 (phy #0): ctrl. port TX status (cookie 5): acked
[2025-07-30 21:27:38.172223]: wlan0 (phy #0): ctrl. port TX status (cookie 6): acked
[2025-07-30 21:27:38.248038]: wlan0 (phy #0): scan started
[2025-07-30 21:27:40.988816]: wlan0 (phy #0): scan finished: 2417 2422 2427 2432 2442 2447 2452 2457 2467 2472, ""
[2025-07-30 21:27:43.587645]: wlan0 (phy #0): scan started
[2025-07-30 21:27:49.702164]: wlan0 (phy #0): scan finished: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825 2412 2417 2422 2427 2432 2437 2442 2447 2452 2457 2462 2467 2472, "0"
[2025-07-30 21:27:50.276990]: wlan0 (phy #0): scan started
[2025-07-30 21:27:52.575504]: wlan0 (phy #0): scan finished: 2412 2437 2462, ""
[2025-07-30 21:27:53.149914]: wlan0 (phy #0): scan started
[2025-07-30 21:27:57.708853]: wlan0 (phy #0): scan finished: 5180 5200 5220 5240 5260 5280 5300 5320 5500 5520 5540 5560 5580 5600 5620 5640 5660 5680 5700 5720 5745 5765 5785 5805 5825, ""
[2025-07-30 21:27:58.282627]: wlan0 (phy #0): scan started
[2025-07-30 21:28:01.005490]: wlan0 (phy #0): scan finished: 2417 2422 2427 2432 2442 2447 2452 2457 2467 2472, ""
[2025-07-30 21:28:10.862046]: wlan0 (phy #0): scan started
```

Beside taking too long to (re)connect, I just notice that scanning also took quite long. Now, I just found and tried my old adapter, which is almost 10y by now and it's pretty quick for scan and connect. From the video, at around 2:40mn, I unplugged comfast adapter and plug my old adapter and it takes only a few second for scanning and connecting. 

https://github.com/user-attachments/assets/12214ae3-da3e-4347-9797-890a94a40302