morrownr/rtw89 #11: 8851bu: sporadically loosing connexion with AP (5GHz)

macmpi, 2025-06-12:
```
[35957.645645] rtw89_8851bu_git 1-2:1.2: rtw89_hw_scan_offload failed ret -110
[35998.907485] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[36040.344439] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[36042.246272] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[36042.454026] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[36085.022135] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[36127.344593] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[36144.515172] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[36144.515207] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[36144.515258] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[36144.515418] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a65
[36144.515467] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515509] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a57
[36144.515554] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a73
[36144.515601] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515656] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a7d
[36144.515698] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515739] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5d
[36144.515778] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a85
[36144.515818] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515859] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515915] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515959] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36144.515998] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a51
[36144.516037] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a65
[36161.212337] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[36161.212375] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[36161.212426] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[36161.212578] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a7b
[36161.212618] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212658] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212698] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212738] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a63
[36161.212778] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212818] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5b
[36161.212877] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212920] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.212962] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a6b
[36161.213002] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.213043] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a55
[36161.213086] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce5
[36161.213134] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36161.213185] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[36165.323062] wlx90de808c6dd9: authenticate with XX:XX:XX:XX:XX:XX (local address=90:de:80:8c:6d:d9)
[36165.323066] wlx90de808c6dd9: send auth to XX:XX:XX:XX:XX:XX (try 1/3)
[36165.324964] wlx90de808c6dd9: authenticated
[36165.325994] wlx90de808c6dd9: associate with XX:XX:XX:XX:XX:XX (try 1/3)
[36165.328875] wlx90de808c6dd9: RX AssocResp from XX:XX:XX:XX:XX:XX (capab=0x911 status=0 aid=3)
[36165.446709] wlx90de808c6dd9: associated
[36165.446796] wlx90de808c6dd9: Limiting TX power to 20 (20 - 0) dBm as advertised by XX:XX:XX:XX:XX:XX
```

macmpi, 2025-06-13:
another instance (while using bluetooth if that matters)
```
[78016.438667] input: Aeropex by AfterShokz (AVRCP) as /devices/virtual/input/input14
[78204.388086] Bluetooth: Unexpected continuation frame (len 0)
[78204.580112] Bluetooth: Unexpected continuation frame (len 0)
[79472.165337] rtw89_8851bu_git 1-2:1.2: rtw89_hw_scan_offload failed ret -110
[79525.414510] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[79560.841124] Bluetooth: Unexpected continuation frame (len 0)
[79561.030331] Bluetooth: Unexpected continuation frame (len 0)
[79581.329621] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[79583.874702] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[79584.079666] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[79634.232365] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[79682.205095] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[79702.334841] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[79702.334881] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[79702.334923] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[79702.335082] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a7f
[79702.335127] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a67
[79702.335173] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[79702.335215] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5f
[79702.335261] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[79702.335309] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[79702.335351] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5f
[79702.335394] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a7b
[79702.335439] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a83
[79702.335507] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a59
[79702.335550] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a15
[79702.335595] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[79702.335641] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a57
[79702.335687] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a75
[79702.335738] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[79721.257907] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[79721.257967] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[79721.258007] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[79721.258399] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f87
[79721.258471] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6b
[79721.258516] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6b
[79721.258587] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79721.258630] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79721.258675] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f83
[79721.259487] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f81
[79721.259542] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79721.259588] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79721.259643] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f75
[79721.259694] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd7
[79721.259783] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79721.259835] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f81
[79721.259886] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6d
[79721.259934] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6b
[79739.400565] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[79739.400607] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[79739.400647] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[79739.400929] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f7d
[79739.400979] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401071] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401121] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f67
[79739.401167] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8904ed7
[79739.401217] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401264] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f73
[79739.401311] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6f
[79739.401360] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd9
[79739.401432] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f89
[79739.401481] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401553] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccdd
[79739.401605] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401676] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[79739.401730] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901faf
[79757.692508] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[79757.692549] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[... 1040 lines cut ...]
[79779.259459] rtw89_8851bu_git 1-2:1.2: read rf busy swsi
[79779.262159] rtw89_8851bu_git 1-2:1.2: read rf busy swsi
[79779.265365] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.268038] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.272290] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.275379] rtw89_8851bu_git 1-2:1.2: read rf busy swsi
[79779.278454] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.282040] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.285899] rtw89_8851bu_git 1-2:1.2: read rf busy swsi
[79779.289210] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.291958] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.294641] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.298635] rtw89_8851bu_git 1-2:1.2: read rf busy swsi
[79779.301292] rtw89_8851bu_git 1-2:1.2: write rf busy swsi
[79779.420688] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[79780.332569] Bluetooth: Unexpected continuation frame (len 0)
[... cut ...]
[79795.259000] wlx90de808c6dd9: authenticate with XX:XX:XX:XX:XX:XX (local address=90:de:80:8c:6d:d9)
[79795.259018] wlx90de808c6dd9: send auth to XX:XX:XX:XX:XX:XX (try 1/3)
[79795.261119] wlx90de808c6dd9: authenticated
[79795.261894] wlx90de808c6dd9: associate with XX:XX:XX:XX:XX:XX (try 1/3)
[79795.265724] wlx90de808c6dd9: RX AssocResp from XX:XX:XX:XX:XX:XX (capab=0x911 status=0 aid=3)
[79795.385945] wlx90de808c6dd9: associated
[79795.386133] wlx90de808c6dd9: Limiting TX power to 20 (20 - 0) dBm as advertised by XX:XX:XX:XX:XX:XX
```

macmpi, 2025-06-15:
> What about your CPU and motherboard?

Intel(R) Core(TM) i3-10110U CPU @ 2.10GHz, Star Labs [LabTop Mk-IV](https://fr.starlabs.systems/collections/parts/products/mainboard-i3-10110u?variant=42093716373758)
So far no errors (need to experiment more), except seemingly more of:
```
[  140.199396] workqueue: delayed_fput hogged CPU for >10000us 4 times, consider switching to WQ_UNBOUND
[  140.211387] workqueue: delayed_fput hogged CPU for >10000us 5 times, consider switching to WQ_UNBOUND
[  166.075400] workqueue: delayed_fput hogged CPU for >10000us 7 times, consider switching to WQ_UNBOUND
[  208.065775] workqueue: delayed_fput hogged CPU for >10000us 11 times, consider switching to WQ_UNBOUND
[  220.704450] Bluetooth: hci0: AOSP extensions version v1.00
[  253.268458] workqueue: delayed_fput hogged CPU for >10000us 19 times, consider switching to WQ_UNBOUND
[  387.196170] workqueue: delayed_fput hogged CPU for >10000us 35 times, consider switching to WQ_UNBOUND
...
[24281.195460] perf: interrupt took too long (2502 > 2500), lowering kernel.perf_event_max_sample_rate to 79000
```
May be linked to other embedded changes related to polling delays?

macmpi, 2025-06-15:
> Didn't you use the usb_storage quirk?

Yes I did:
```
$ sudo dmesg | grep "usb-storage 1-2:1.0: device ignored"
[    1.807545] usb-storage 1-2:1.0: device ignored
[  976.945333] usb-storage 1-2:1.0: device ignored 
```
(2nd instance is when I removed/reinserted dongle to check if occurrence of `delayed_fput hogged CPU` was linked to dongle: seems it is)

dubhater, 2025-06-15:
By the way, are you using iwd?

macmpi, 2025-06-17:
is https://github.com/morrownr/rtw89/commit/7595933582a26a29d7ef2ce1a0ef7c12fca8c8d5 supposed to address this issue or is it for sth else?

dubhater, 2025-06-17:
Only if you were seeing the message "avoiding multiple of 512" when the connection was dying.

dubhater, 2025-06-17:
No fix yet, but I have another question: did you install the wireless-regdb package and set the country code? What does ``iw reg get`` say?

dubhater, 2025-06-17:
I hope it does.

macmpi, 2025-06-27:
now on https://github.com/morrownr/rtw89/commit/243fe83fdceec4d4cba96a8ce66b4a27c7cf4d6a, and
`6.11.0-28-generic #28~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Fri May 23 10:31:01 UTC 2 x86_64 x86_64 x86_64 GNU/Linux`
found this on `dmesg` (proper country set). wifi still working tough 
```
[22021.495667] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[22021.701346] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[22023.300905] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[22024.563976] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[22025.605109] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[22025.605178] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[22025.605238] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[22025.605570] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a57
[22025.605638] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5b
[22025.605703] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.605770] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.605835] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a81
[22025.605901] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.605965] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606031] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5f
[22025.606099] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606187] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606255] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606321] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a83
[22025.606384] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606475] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[22025.606549] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a5f

```

a5a5aa555oo, 2025-06-29:
@dubhater 

After applying the patch, My pc was unable to find any 5GHz SSIDs in step 3.

[image]

[dmesg.txt](https://github.com/user-attachments/files/20970663/dmesg.txt)

a5a5aa555oo, 2025-06-30:
Tested for about 4 hours, no disconnection and error occured.

[dmesg_4H.txt](https://github.com/user-attachments/files/20972605/dmesg_4H.txt)

dubhater, 2025-06-30:
@a5a5aa555oo Thank you. Not sure why PCI is okay. I poked and prodded it a bit more and it still looks like a firmware bug.

dubhater, 2025-06-30:
@a5a5aa555oo Actually, could you test with firmware 0.29.29.8 as well? 1.29.29.9 works here, but 0.29.29.8 doesn't.

a5a5aa555oo, 2025-06-30:
@dubhater 

With 8852B firmware v0.29.29.8, I got some errors. Interestingly Wi-Fi is still working.

[dmesg_fw_v0.29.29.8.txt](https://github.com/user-attachments/files/20986822/dmesg_fw_v0.29.29.8.txt)

dubhater, 2025-06-30:
Yay, a firmware bug!

dubhater, 2025-06-30:
@macmpi It's kind of random. It depends on the number of DFS/no IR channels, which depends on your country code but also on the hints picked up from the beacons transmitted by the networks around you.

Anyway, if you update the driver now you should not see the problem anymore.

macmpi, 2025-07-01:
@dubhater thanks for debunking this!
Hopefully firmware will be fixed at some point, even if driver now work around this.
Installed with https://github.com/morrownr/rtw89/commit/19937f1b20df7b55adb8a588e66c1fe3e345b5c2 and will keep monitoring for a while.
Closing for now.
Thanks: hope it will make it for 6.17 !

macmpi, 2025-07-07:
hum... look's like it's still there unfortunately (or similar):
```
[ 9357.577859] warning: `Socket Thread' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[10622.172685] rtw89_8851bu_git 1-2:1.2: rtw89_hw_scan_offload failed ret -110
[10623.353476] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[10624.608818] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[10625.600475] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[10625.806392] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[10627.321256] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[10628.495271] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[10629.486452] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[10629.486492] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[10629.486541] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[10629.486702] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a55
[10629.486743] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900ad5
[10629.486798] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a6f
[10629.486852] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a7f
[10629.486901] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a6f
[10629.486977] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[10629.487025] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb89088e5
[10629.487094] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a79
[10629.487138] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a85
[10629.487181] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce1
[10629.487226] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a6d
[10629.487275] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a85
[10629.487318] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890cce3
[10629.487362] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a59
[10629.487405] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8900a6d
[10630.020419] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[10630.020512] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[10630.020603] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[10630.020764] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6d
[10630.020818] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.020872] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f7f
[10630.020925] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f67
[10630.020969] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f7f
[10630.021018] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6f
[10630.021064] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f69
[10630.021109] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f83
[10630.021157] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6f
[10630.021208] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.021254] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd9
[10630.021306] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6d
[10630.021351] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6b
[10630.021394] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f71
[10630.021446] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6f
[10630.554695] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[10630.554733] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[10630.554795] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[10630.555147] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555188] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555228] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555268] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f75
[10630.555308] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f71
[10630.555352] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555395] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555437] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555480] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f8d
[10630.555520] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f6f
[10630.555560] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f67
[10630.555603] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555644] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10630.555695] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f81
[10630.555740] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb890ccd5
[10631.012810] rtw89_8851bu_git 1-2:1.2: [ERR]FWDL path ready
[10631.012879] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x1E0 = 0x23
[10631.012944] rtw89_8851bu_git 1-2:1.2: [ERR]fwdl 0x83F0 = 0x70000
[10631.013210] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f8b
[10631.013278] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f7f
[10631.013349] rtw89_8851bu_git 1-2:1.2: [ERR]fw PC = 0xb8901f75
[... 2249 lines cut ...]
[10735.124482] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10736.227404] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10737.331374] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10738.435365] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10739.539261] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10740.643319] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10741.747059] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10742.851109] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10743.954878] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10745.058835] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10746.162778] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10747.266625] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10748.370502] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[... cut ...]
[10752.802248] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10753.514148] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10756.777995] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10761.777684] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10768.776944] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10773.250635] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10778.776236] rtw89_8851bu_git 1-2:1.2: Update probe request failed
[10780.280208] rtw89_8851bu_git 1-2:1.2: Update probe request failed
```

dubhater, 2025-07-07:
Okay, then please update the driver and load it with debug_mask=0x8000. Make sure the new messages are appearing in dmesg (like a big hex dump). When the problem happens again attach the output of dmesg.

a5a5aa555oo, 2025-07-07:
@macmpi 

You can simply modify the line 10 in /etc/modprobe.d/rtw89.conf and then reload the rtw89 drivers.

macmpi, 2025-07-07:
ok, seeing the messages now. Will let go until next issue and post

dubhater, 2025-07-08:
That's okay. Those are not important.

macmpi, 2025-07-26:
funny enough, just got it!
```
[337013.951626] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[337015.221619] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[337016.494674] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[337017.766129] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[337019.000191] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
[337020.145352] rtw89_8851bu_git 1-2:1.2: c2h reg timeout
```

dubhater, 2025-07-26:
Please attach the entire output of dmesg.

macmpi, 2025-07-26:
here you go [log.txt.tar.gz](https://github.com/user-attachments/files/21445043/log.txt.tar.gz)

dubhater, 2025-07-26:
Ah, sorry, the debugging messages filled the kernel's buffer and pushed out the interesting part. Please attach the output of ``journalctl -b`` instead.

macmpi, 2025-07-26:
compressed file is about 1.8Mb... shall I send by email instead?

dubhater, 2025-07-26:
Well, it's a problem with the hardware scan feature. Maybe the same one from before but triggered in a different way. We're still waiting for a fix from Realtek.

dubhater, 2025-07-26:
Sure. I will see it when they release a new firmware.

a5a5aa555oo, 2025-08-02:
I got this sometimes, no `rtw89_hw_scan_offload failed ret -110` in the kernel log.

[8851bu.txt](https://github.com/user-attachments/files/21559017/8851bu.txt)

macmpi, 2025-08-28:
> Sure. I will see it when they release a new firmware.

seems like they recently fixed (maybe) similar problems for `8852b/c`: hopefully `8851bu` will come soon...
https://gitlab.com/kernel-firmware/linux-firmware/-/commits/main/rtw89?ref_type=heads

a5a5aa555oo, 2025-09-01:
> I got this sometimes, no `rtw89_hw_scan_offload failed ret -110` in the kernel log.
> 
> [8851bu.txt](https://github.com/user-attachments/files/21559017/8851bu.txt)

This happens in AP mode too. It always dies silently, no helpful messages found in the kernel log.

kernel log (AP side)
[dmesg_ap.txt](https://github.com/user-attachments/files/22082907/dmesg_ap.txt)

kernel log ( client side)
[dmesg_client.txt](https://github.com/user-attachments/files/22082910/dmesg_client.txt)

hostapd config file
[8851bu_5g.conf.tar.gz](https://github.com/user-attachments/files/22082828/8851bu_5g.conf.tar.gz)

a5a5aa555oo, 2025-09-02:
@dubhater 

Thank you! When it disconnects next time, I'll apply your patch and keep testing.

a5a5aa555oo, 2025-09-19:
My 8851BU dongle has been running as an AP for over 7 days and it is still alive, so the driver should have no problem. 👍

<details>
<summary>kernel log</summary>
<pre><code>
[Sep11 04:00] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[  +6.585929] rtw89_8851bu_git 1-2:1.2: loaded firmware rtw89/rtw8851b_fw.bin
[  +0.000310] rtw89_8851bu_git 1-2:1.2: git commit ab336292de1efe716dbb701dfa3bcea0617117b7
[  +0.054042] rtw89_8851bu_git 1-2:1.2: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 5
[  +0.000022] rtw89_8851bu_git 1-2:1.2: Firmware version 0.29.41.3 (65cefb31), cmd version 0, type 3
[  +1.031606] rtw89_8851bu_git 1-2:1.2: chip rfe_type is 4
[  +0.136745] rtw89_8851bu_git 1-2:1.2: rfkill hardware state changed to enable
[  +0.000169] usbcore: registered new interface driver rtw89_8851bu_git
[  +0.002811] rtw89_8851bu_git 1-2:1.2 wlxdceae7ad196c: renamed from wlan0
[ +14.141474] br0: port 3(wlxdceae7ad196c) entered blocking state
[  +0.000017] br0: port 3(wlxdceae7ad196c) entered disabled state
[  +0.000027] rtw89_8851bu_git 1-2:1.2 wlxdceae7ad196c: entered allmulticast mode
[  +0.000130] rtw89_8851bu_git 1-2:1.2 wlxdceae7ad196c: entered promiscuous mode
[  +0.000057] br0: port 3(wlxdceae7ad196c) entered blocking state
[  +0.000008] br0: port 3(wlxdceae7ad196c) entered forwarding state
[  +0.001810] br0: port 3(wlxdceae7ad196c) entered disabled state
[Sep11 04:01] br0: port 3(wlxdceae7ad196c) entered blocking state
[  +0.000020] br0: port 3(wlxdceae7ad196c) entered forwarding state
[  +9.557172] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[  +0.250051] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[Sep11 04:18] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 04:35] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 05:38] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 05:56] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 05:57] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 06:21] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 07:12] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 07:20] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 07:46] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 07:58] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:02] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:03] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[ +34.898144] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:04] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[  +0.205472] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:06] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:16] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:17] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:21] perf: interrupt took too long (2596 > 2500), lowering kernel.perf_event_max_sample_rate to 77000
[Sep11 08:22] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:23] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:30] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[  +0.205356] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:40] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:50] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[ +17.520240] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[  +0.241888] rtw89_8851bu_git 1-2:1.2: timed out to flush queues
[Sep11 08:51] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[  +5.326568] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[  +4.606867] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:53] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[ +23.965796] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 08:55] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:02] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:03] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:14] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:25] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:26] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:27] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:28] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:41] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:47] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 09:57] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 10:05] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep11 10:10] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[... cut ...]
[Sep19 18:49] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep19 19:07] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep19 19:29] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep19 19:31] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit
[Sep19 19:39] usb 3-1.3: rtl8xxxu_send_beacon_frame: Failed to read beacon valid bit

#### Both 2.4G AP and 5G AP are still alive although there are many harmless messages in the kernel log.
</code></pre>
</details>

dubhater, 2025-10-15:
> SSID is still visible and can be associated, but no response received when i pinged it.

That's interesting. So it can transmit and receive management frames. I assume you use encryption on that AP. Does the WPA handshake happen? Those are data frames but they go through a higher priority queue compared to ping (I think).

a5a5aa555oo, 2025-10-15:
@dubhater 

My AP PC:
1. Noname 8822BU dongle ==> 5G AP
2. D-Link AX9U ==> 2.4G AP

My Client PC:
Wi-Fi card: 8852CE (driver: rtw89_8852ce in kernel 6.12.50-1-lts)

I just restarted my client pc, and it's weird that 2.4G connection revived. I can receive responses from the AP now. -_-
Actually when qbittorrent stopped transferring, the driver for 8852CE (rtw89_8852ce) crashed too. maybe this is the real cause?

a5a5aa555oo, 2025-10-16:
AP PC:
Noname 8822BU dongle ==> 5G AP
D-Link AX9U ==> 2.4G AP

Client PC:
Wi-Fi card: 8852CE connected to the 2.4G AP (driver: rtw89_8852ce in kernel 6.12.50-1-lts)

This morning the 8852CE driver crashed, but everything works fine after reloading the 8852CE driver, unable to reproduce the problem happened yesterday.

Summary:
10/16 22:00: qBittorrent started up right after boot.
10/17 02:38: 8852CE driver crashed. qBittorrent stopped transferring.
10/17 04:39: I woke up, NetworkManager showed client PC was still associated with the AP, but network was unavailable actually. Everything works fine after reloading the 8852CE driver.

The 8852CE driver was added into the kernel earlier than the 8852BE driver, i cannot believe it is much more unstable than the 8852BE driver and crashed in such a  light traffic. 🙄

kernel log (AP)
[dmesg_AP.txt](https://github.com/user-attachments/files/22959621/dmesg_AP.txt)

kernel log (client)
[dmesg_client.txt](https://github.com/user-attachments/files/22959642/dmesg_client.txt)

dubhater, 2025-10-23:
@F-Poole Is the wireless-regdb package installed and the country code set correctly? See with ``iw reg get``.

F-Poole, 2025-10-28:
I've installed wireless-regdb with the correct country code and, in the last days, I've not lost the connection but I've not used it much either.