morrownr/rtw89 #12: Require Wifi and Bluetooth drivers for 6.15.2 kernel

Blowingmemes, 2025-06-13:
I want Wifi and Bluetooth Driver for TPLink Archer TX10UB Nano
Bus 001 Device 016: ID 3625:010b Realtek 802.11ax WLAN Adapter
Kernel version : 6.15.2-arch1-1
Distro: Arch Linux

Blowingmemes, 2025-06-13:
Alright Thanks. What about bluetooth driver?

Blowingmemes, 2025-06-13:
Arch Linux still  freezes whenever the usb wifi adapter is being utilized.

dubhater, 2025-06-13:
Next time it happens run ``sudo dmesg > dmesg.txt`` and attach dmesg.txt here.

Blowingmemes, 2025-06-14:
[dmesg.txt](https://github.com/user-attachments/files/20737724/dmesg.txt)

a5a5aa555oo, 2025-06-14:
> [   14.524217] module 8851bu: .gnu.linkonce.this_module section size must match the kernel's built struct module size at run time

@dubhater 

It looks like... @Blowingmemes's system has both vendor driver and rtw89 driver installed, right? 🤔

dubhater, 2025-06-14:
Yes, it looks that way.

a5a5aa555oo, 2025-06-14:
@Blowingmemes 

Regarding to bluetooth driver, I think [this one](https://github.com/a5a5aa555oo/realtek_bluetooth_ubuntu_6.8) should work.

Steps to install it:
```
git clone https://github.com/a5a5aa555oo/realtek_bluetooth_ubuntu_6.8
```
```
cd realtek_bluetooth_ubuntu_6.8
```
```
sudo dkms install $PWD
```

dubhater, 2025-06-14:
> If your dongle still doesn't work, run sudo dmesg > dmesg.txt and attach dmesg.txt to this issue again so that @dubhater can take a look what is wrong.

But first please update this driver again. I pushed something just now.

Blowingmemes, 2025-06-16:
Unforturnately it didn't help at all. I reinstalled arch linux and the driver aswell but the issue still exists. The adapter connects to wifi and works for few minutes and then it randomly disconnects and turns off the wifi. The only way to turn it back on again is just to plugging off and on but its useless to do it if the problem still exists.

[dmesg.txt](https://github.com/user-attachments/files/20749739/dmesg.txt)

dubhater, 2025-06-16:
So the same driver (from this repository) works in Debian on the same computer?

> [   44.373420] platform regulatory.0: Direct firmware load for regulatory.db failed with error -2

Try to install the wireless-regdb package and uncomment the line with your country code in /etc/conf.d/wireless-regdom. Then reboot.

Blowingmemes, 2025-06-18:
> So the same driver (from this repository) works in Debian on the same computer?
>
>
> Try to install the wireless-regdb package and uncomment the line with your country code in [...]

This absolutely fixed the problem. Thank you so much

closed 2025-06-19 by Blowingmemes

dubhater, 2025-06-30:
It should work even without the wireless-regdb package now.