morrownr/rtw89 #15: rtl8852/32cu : test results

morrownr, 2025-07-04:
@dubhater 

Code is up to date as of your patches of July 03.

Firmware in use: 0.27.129.1

Ubuntu 25.04, kernel 6.14

Issue: Not long, maybe 20 minutes, after updating to firmware 0.27.129.1, the wireless interface crashed and was no longer available. There was no recovery. A reboot was required.

Log:

```
[ 1636.176624] rtw89_8852cu_git 5-2.3:1.0: timed out to flush queues
[ 1636.381130] rtw89_8852cu_git 5-2.3:1.0: timed out to flush queues
[ 1636.590704] rtw89_8852cu_git 5-2.3:1.0: timed out to flush queues
```

Will continue testing.

dubhater, 2025-07-05:
It's interesting that it didn't reconnect. Was there nothing else around those messages?

morrownr, 2025-07-05:
@dubhater 

> It's interesting that it didn't reconnect. Was there nothing else around those messages?

Nothing. I did a reboot and the interface came up and connected. I am continuing to test.

Here is where I am today:

$ iperf3 -c 192.168.1.1 -R -t 3000

```
...
[  5] 2935.00-2936.00 sec  39.1 MBytes   328 Mbits/sec                  
[  5] 2936.00-2937.00 sec  38.8 MBytes   325 Mbits/sec                  
[  5] 2936.00-2937.00 sec  38.8 MBytes   325 Mbits/sec                  
iperf3: error - control socket has closed unexpectedly
```

Interestingly it made it to almost 3000 seconds. Nothing in the log. During this command, I had to be away from the test system and it went into suspend, after moving the mouse the system came back from suspend, iperf3 showed 0 Mbits/sec for about 12 seconds and then the connection resumed as one would expect.

Here is where I tell you what you already know:

- USB3 is not working
- LED is not working

I am continuing to test managed mode. Hopefully I can find something that might be of help.

dubhater, 2025-07-09:
@Kiwimannen Are you testing with firmware 0.27.129.1 ? It showed some promise here: no disconnection in 28 hours, until I broke the uptime to investigate a different bug.

dubhater, 2025-07-09:
@Kiwimannen Awesome, I will keep an eye out.

@morrownr Yes, that's a good idea.

morrownr, 2025-07-09:
@Kiwimannen 

> initial problems (cd-rom mode

During the time I worked on bringing up the 8852cu out-of-kernel driver, it became clear to me that this particular chipset seems to have an above average amount of problems with storage mode. The problems can vary between computer systems. I have 2 systems here that are problematic and one system that is not problematic. What symptoms are you seeing?

Kiwimannen, 2025-07-10:
I know about iperf3, but I have never used it. I think I can even install on my OpenWRT router ?

It was just for fun, but unfortunatly my Ubuntu installation on risc-v is lacking both kernel-headers (and cant install any for some reason) and a build folder under /lib/modules/6.6.67-ky/  (Its an Orange Pi RV2 BTW ) so I cant install rtw89 at the moment.

morrownr, 2025-07-10:
@Kiwimannen 

> I know about iperf3, but I have never used it. I think I can even install on my OpenWRT router ?

Yes, it is available to install in OpenWRT. The package name is `iperf3`. In fact, having iperf3 running on your wifi router is optimal. Do you use Luci?

morrownr, 2025-07-16:
USB3 switching is working fine so far. Thanks.

I have a plan to go deeper into the 3 issues that I reported. My plan is to work them in the following order unless you would prefer something different. Does not matter to me.

Unable to use WPA3, WPA2 works fine.
Unable to activate WiFi 6. WiFi 5 is working fine.
Unable to active 160 MHz channel width, 80 works fine.

morrownr, 2025-07-18:
@Kiwimannen 

> I usually use OpenWRT for AP and Ubuntu only has a gui version in the settings that does not allow you to choose individual radios - but I can look it up and probably set up a manual AP with settings and use 6GHz somehow tomorrow.

I use OpenWRT but it is not a good platform for testing drivers for several reasons. Do you have a Raspberry Pi? I have an AP guide up in the site Main Menu. I can help. 

> First time I used iperf3 so I dont kow what results to expect but I will test it against a different router tomorrow and see if the results differ.

Unless there is a load that starts on the channel, I generally see more stable numbers.

morrownr, 2025-07-19:
Short update so that we can compare notes:

I am still testing AP mode using hostapd v2.10 with the rtl8852cu driver here in this rtw89 repo.
 
AP mode is far more complicated to deal with than is managed mode so it can take some time. I have narrowed my concerns with AP mode down to two items:

1. Security

WPA2 and WPA3 Transitional work fine. WPA3 is not working. I need to do a deep dive to see where things are going wrong. Interesting, in WPA3 Transitional mode, if a client connects, it will show WPA3 connection. There is a little something wrong somewhere but I do not know what at this point. WPA3 works with the vendor driver.

2. 160/ 80+80 MHz channel width
 
Unable to successfully test 160 / 80+80 channel width in AP mode. This based on 5 GHz as the regdb prevents me from testing 6 GHz AP mode in my country. I was able to test the vendor driver successfully at 160 MHz channel width but it has been a few months ago. I need to install the vendor driver and do some comparisons to see if I can figure this out.

Other than those 2 items, AP mode seems to be working well. Need testers for 6 GHz AP mode. Users from EU countries should have no problems doing 6 GHz AP mode tests.

Last note: I am also going to mix in AP mode testing of the rtl8852bu driver from here in this rtw89 repo. 

Equipment status:

Thanks to @Kiwimannen , I have a rtl8912au based adapter on the way. That means the only Realtek WiFi 6/7 based USB adapter in common use that I am missing is the rtl8851/33bu. After talking to @dubhater , there is an Edimax adapter that is single-state that I will get when it is available at one of the places that I show.

I think the only adapter that @dubhater is missing, once he gets the rtl8912au based adapter, is a rtl8852/32au based adapter.

dubhater, 2025-07-20:
That's right, I'm only missing RTL8832AU.

morrownr, 2025-07-20:
@dubhater 

> That's right, I'm only missing RTL8832AU.

I'm still working this issue.

> 160/ 80+80 MHz channel width

Remember: AP mode, hostapd, rtl8832cu based adapter, rtw89 driver in this repo.

Status update and a question:

The question is: does this driver support 80+80 in AP mode?

Here is what causes the question: DFS channels with USB WiFi adapters in AP mode is a non starter as support for processing and taking action is not available unless you see something that I do not. That leaves regular 6 GHz channels as the only option for 160 MHz channel width in my country unless 80+80 is supported. In my country, the only option for 80+80 is channels 36-48 (one 80MHz channel) and 149-161 (another 80MHz channel). I cannot seem to make 80+80 work and I suspect the capability is simply not in the code. It would not be necessary for band 4, 6 GHz, so 80+80 may simply not be supported which means I am wasting my time.

I said earlier that the vendor driver previously did support 160 channel width but, after checking my notes, I need to withdraw that statement as I had modified the vendor driver to ...

So, pending your thoughts on 80/80 support for 5 GHz, I am down to the WPA3 issue. I do have some cosmetic issues. Here is an example:

This driver shows txpower of 23 when on 5 GHz on channel 36. That is correct for my country. However, the vendor driver shows 12, which is not correct for my country but testing shows that the adapter is actually outputting a stronger signal with the vendor driver. What we want is correct txpower output and the better results.

dubhater, 2025-07-20:
rtw89 only advertises support for 160 MHz. I saw some code related to 80+80, so maybe they can enable it if you ask.

The vendor driver just reports 12 dbm all the time. The number is hardcoded.

rtw89 doesn't report the TX power to iw at all but we can see something in debugfs: ``cat /sys/kernel/debug/ieee80211/phy*/rtw89/txpwr_table``. According to that, Brostrend wanted the TX power to be 18 dbm:

```
[TX power byrate]                                                                    
CCK             -  1M     2M     5.5M   11M     |  20,  20,  20,  20,   dBm
LEGACY          -  6M     9M     12M    18M     |  18,  18,  18,  18,   dBm
LEGACY          -  24M    36M    48M    54M     |  18,  18,  17,  16,   dBm
MCS_1NSS        -  MCS0   MCS1   MCS2   MCS3    |  18,  18,  18,  18,   dBm
MCS_1NSS        -  MCS4   MCS5   MCS6   MCS7    |  18,  17,  16,  15,   dBm
MCS_1NSS        -  MCS8   MCS9   MCS10  MCS11   |  14,  13,  12,  11,   dBm
HEDCM_1NSS      -  MCS0   MCS1   MCS3   MCS4    |  18,  18,  18,  18,   dBm
MCS_2NSS        -  MCS0   MCS1   MCS2   MCS3    |  18,  18,  18,  18,   dBm
MCS_2NSS        -  MCS4   MCS5   MCS6   MCS7    |  18,  17,  16,  15,   dBm
MCS_2NSS        -  MCS8   MCS9   MCS10  MCS11   |  14,  13,  12,  11,   dBm
HEDCM_2NSS      -  MCS0   MCS1   MCS3   MCS4    |  18,  18,  18,  18,   dBm
```

But rtw89 is limiting it:

```
[TX power limit]   
CCK_1TX_20M     -  NON_BF  BF   |   0,   0,             dBm                                                                                                                
CCK_1TX_40M     -  NON_BF  BF   |   0,   0,             dBm                                                                                                                
OFDM_1TX        -  NON_BF  BF   |  16,   0,             dBm                                                                                                                
MCS_1TX_20M_0   -  NON_BF  BF   |  16,   0,             dBm                                                                                                                
MCS_1TX_20M_1   -  NON_BF  BF   |  16,   0,             dBm                                                                                                                
MCS_1TX_20M_2   -  NON_BF  BF   |  16,   0,             dBm                                                                                                                
MCS_1TX_20M_3   -  NON_BF  BF   |  16,   0,             dBm                                                                                                                
MCS_1TX_20M_4   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_20M_5   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_20M_6   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_20M_7   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_40M_0   -  NON_BF  BF   |  16,   0,             dBm
MCS_1TX_40M_1   -  NON_BF  BF   |  16,   0,             dBm
MCS_1TX_40M_2   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_40M_3   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_80M_0   -  NON_BF  BF   |  15,   0,             dBm
MCS_1TX_80M_1   -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_160M    -  NON_BF  BF   |   0,   0,             dBm
MCS_1TX_40M_0p5 -  NON_BF  BF   |  16,   0,             dBm
MCS_1TX_40M_2p5 -  NON_BF  BF   |   0,   0,             dBm
CCK_2TX_20M     -  NON_BF  BF   |   0,   0,             dBm
CCK_2TX_40M     -  NON_BF  BF   |   0,   0,             dBm
OFDM_2TX        -  NON_BF  BF   |  13,   0,             dBm
MCS_2TX_20M_0   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_20M_1   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_20M_2   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_20M_3   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_20M_4   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_20M_5   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_20M_6   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_20M_7   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_40M_0   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_40M_1   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_40M_2   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_40M_3   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_80M_0   -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_80M_1   -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_160M    -  NON_BF  BF   |   0,   0,             dBm
MCS_2TX_40M_0p5 -  NON_BF  BF   |  13,  10,             dBm
MCS_2TX_40M_2p5 -  NON_BF  BF   |   0,   0,             dBm
```

That's as far as I looked.

Kiwimannen, 2025-07-20:
> ignore the examples in the guide and start with the hostapd-rtl8852cu.conf file located in the repo with the vendor driver:
> 
> https://github.com/morrownr/rtl8852cu-20240510
> 
> This is a project you can have some fun with and help with a lot of testing.

Everything but the hostapd.conf file seems to be working for me (rasp4b, rtw89, rtl8832cu) I used the one above and made all the appropriate changes for running AP with 6GHz 160MHz - but it wont start.

Can someone post a short version of a working hostapd.conf for 6GHz with just the necessary commands ?

morrownr, 2025-07-21:
> Can someone post a short version of a working hostapd.conf for 6GHz with just the necessary commands ?

Post the file that you have as is and I will suggest what to modify.

Kiwimannen, 2025-07-21:
> Post the file that you have as is and I will suggest what to modify

hostapd-6GHz.conf

``# SSID
ssid=rtl8832cu_6GHz

# PASSPHRASE
wpa_passphrase=Bapelsin4014

# Band
# a = 5 GHz & 6 GHz (a/n/ac/ax), g = 2 Ghz (b/g/n)
hw_mode=a

# Channel
channel=65

# Operating Class

# 134 signifies channel width: 160 MHz.
# Channel (Center Frequency - he_oper_centr_freq_seg0_idx)
# Lower: U-NII-5
# 1 (15), 33(47), 65(79)
# Upper: U-NII-6, 7, 8
# 97(111), 129(143), 161(175), 193(207)
# PCS Channel Numbers
# 5/21, 37/53, 69/85, 101/117, 133/149, 165/181, 197/213
#
# Activate a line below with desired setting for 6 GHz operation.
op_class=134

# Country code (ISO/IEC 3166-1). Used to set regulatory domain.
country_code=SE
#
# This parameter is used to set the third octet of the country string.
# All environments of the current frequency band and country (default)
country3=0x20

# Advertises the country code, allowed channels and transmit power.
ieee80211d=1
# Enables support for 5 GHz DFS channels if supported.
# Requires ieee80211d=1 (above)
ieee80211h=1

# Interfaces
# Bridge interface
bridge=br0
#wds_sta=1
# WiFi interface (wlan0 should be changed to your interface name)
interface=wlan0

# Set hostapd driver (nl80211 is used with all Linux mac80211 (in-kernel)
# and modern Realtek drivers)
driver=nl80211

# Event logger configuration
logger_syslog=-1
logger_syslog_level=2

# Control interface - (i.e. for hostapd_cli)
ctrl_interface=/var/run/hostapd
ctrl_interface_group=0

# Various settings
beacon_int=100
dtim_period=2
skip_inactivity_poll=1

# Security
# authentication algorithms (3 = both are allowed, use 3 for SAE otherwise use 1)
auth_algs=1
# Limit number of concurrent "clients" (stations) and do not restrict them by MAC address
max_num_sta=16
# Station MAC address based authentication
macaddr_acl=0
# hidden SSID (1 = hidden)
ignore_broadcast_ssid=0
# Opportunistic Key Caching (1 = enabled)
okc=1
# Enable WPA. (2 is required for WPA2, mixed and WPA3)
wpa=2
# Pairwise cipher
#wpa_pairwise=CCMP
rsn_pairwise=CCMP CCMP-256 GCMP GCMP-256
#
# Activate only one of the following 3 sections
#
# 3. WPA3 Personal (required for band 4, 6 GHz)
# Some older devices may not be compatible with WPA3.
wpa_key_mgmt=SAE
ieee80211w=2
sae_pwe=2

# Wireless Multimedia Extensions (WME)
# The below line is mandatory for 6 GHz.
wme_enabled=1
# Wi-Fi Multimedia (WMM)
# The below line is mandatory for WiFi 4 and above.
wmm_enabled=1

# IEEE 802.11ax (WiFi 6) configuration
ieee80211ax=1
# Channel width (0 = 40 MHz, 1 = 80 Mhz, 2 = 160 MHz)
# For WiFi 6 on band 1 (2.4 GHz) comment out the below line
he_oper_chwidth=1
# HE center channel (chan + 2 for 40 MHz) (chan + 6 for 80 MHz) (chan + 14 for 160 MHz)
# For WiFi 6 on band 1 (2.4 GHz) comment out the below line
he_oper_centr_freq_seg0_idx=42
#
# Maximum MPDU Length of HE 6 GHz band capabilities.
he_6ghz_max_mpdu=0
#
# Maximum A-MPDU Length Exponent of HE 6 GHz band capabilities. Indicates
he_6ghz_max_ampdu_len_exp=2
#     (default)
he_6ghz_rx_ant_pat=0
#
# Tx Antenna Pattern Consistency of HE 6 GHz capability.
#     (default)
he_6ghz_tx_ant_pat=0
#
# 6 GHz Access Point type
2 = Very low power AP (default)

# This has no impact for operation on other bands.
he_6ghz_reg_pwr_type=0
#
# used by clients to discern the source of interference
# each AP in your area needs to use a different number
# allowed: 1-63
he_bss_color=37
#
# Activate beamforming capabilities
#he_su_beamformee=1
#
# Basic NSS/MCS set
he_mu_edca_qos_info_param_count=0
he_mu_edca_qos_info_q_ack=0
he_mu_edca_qos_info_queue_request=0
he_mu_edca_qos_info_txop_request=0
he_mu_edca_ac_be_aifsn=8
he_mu_edca_ac_be_aci=0
he_mu_edca_ac_be_ecwmin=9
he_mu_edca_ac_be_ecwmax=10
he_mu_edca_ac_be_timer=255
he_mu_edca_ac_bk_aifsn=15
he_mu_edca_ac_bk_aci=1
he_mu_edca_ac_bk_ecwmin=9
he_mu_edca_ac_bk_ecwmax=10
he_mu_edca_ac_bk_timer=255
he_mu_edca_ac_vi_ecwmin=5
he_mu_edca_ac_vi_ecwmax=7
he_mu_edca_ac_vi_aifsn=5
he_mu_edca_ac_vi_aci=2
he_mu_edca_ac_vi_timer=255
he_mu_edca_ac_vo_aifsn=5
he_mu_edca_ac_vo_aci=3
he_mu_edca_ac_vo_ecwmin=5
he_mu_edca_ac_vo_ecwmax=7
he_mu_edca_ac_vo_timer=255

rtw_he_enable=2
rtw_vht_enable=2

rtw_switch_usb_mode=1 

# End``

Was unsure if whats under```
 # Operating Class

# 134 signifies channel width: 160 MHz.
```
was supposed to be behind # but I have tried both.

I get:  $DAEMON_CONF (code=exited, status=1/FAILURE) when running "systemctl status hostapd"

morrownr, 2025-07-21:
Change the following line:

> he_oper_centr_freq_seg0_idx=42

to:

he_oper_centr_freq_seg0_idx=79

Totally remove the following lines:

rtw_he_enable=2
rtw_vht_enable=2

rtw_switch_usb_mode=1

Those lines only apply to the vendor driver, not this rtw89 driver and the lines to not go in the hostapd.conf file anyway. Those lines will screw up the startup of hostapd.

The follow line probably is not necessary.

country3=0x20

Last thoughts: hostapd is excellent in that it gives you the capability to get maximum performance from AP mode. hostapd is actually in OpenWRT. In the case we have here, we are setting things up manually and this gets complicated. I am doing the best I can to help but my warning is this: I have many years experience using hostapd but I have little experience with 6 GHz because the wireless regdb shuts USB WiFi adapter use down on 6 GHz for my country.

Kiwimannen, 2025-07-21:
Thanks, I made those changes but now it says:

`/etc/systemd/system/hostapd.service:6: Unknown key 'Environment' in section [Unit], ignoring`

Do you have any idea what it can mean ?

My SSID is not showing up in any scans.

morrownr, 2025-07-21:
> Do you have any idea what it can mean ?

Yes, run:

sudo nano /etc/systemd/system/hostapd.service

or if you prefer a gui text editor, run:

sudo geany /etc/systemd/system/hostapd.service

Post the contents of the file so that I can see the part that needs to be corrected.

Kiwimannen, 2025-07-21:
That fixed it, thanks :) Hostapd is starting without problems - however, Im still not seeing the SSID when I scan.

Turns out, my wlan0 has been renamed:

`systemd-networkd[314]: wlan0: Interface name change detected, renamed to wlx90de8017a05f`.

Thats because I still have to remove and insert my usb-adapter after every restart to go from cd-rom mode to wifi mode.

Gonna try my second rtl8832cu adapter, but Im pretty sure its gonna behave the same way.

dubhater, 2025-07-21:
> Thats because I still have to remove and insert my usb-adapter after every restart to go from cd-rom mode to wifi mode.
> 
> Gonna try my second rtl8832cu adapter, but Im pretty sure its gonna behave the same way.

You can add this to your kernel's command line: ``usb_storage.quirks=0bda:1a2b:i`` Then usb_modeswitch should be able to switch the adapter without unplugging it.

Kiwimannen, 2025-07-22:
Thanks, both of you - the kernel command line trick works perfect.

I have changed wlan0 to my adapter name wlx90de8017a05f in hostapd.conf, but, unfortunately - it still does not work.

Hostapd seems to be running OK when I check with " systemctl status hostapd", but if I check with iw dev, I get:

```
iw dev
phy#1
	Interface wlx90de8043767f
		ifindex 4
		wdev 0x100000001
		addr 90:de:80:43:76:7f
		type managed
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	0	0	0	0	0	0		0
```
The adapter is still in managed mode.

dubhater, 2025-07-22:
> [@dubhater](https://github.com/dubhater)
> 
> > The vendor driver just reports 12 dbm all the time. The number is hardcoded.
> 
> > rtw89 doesn't report the TX power to iw...
> 
> Dp you know why not providing the correct txpower to iw seems to be the norm?
> 

I don't know. Because it's optional, maybe?

> The below line is showing many times in dmesg. This seems to have shown up once I started testing AP mode. What is this?
> 
> [32186.235349] rtw89_8852cu_git 2-2:1.0: MAC c2h class 0 func 6 not support

The firmware is sending a message which rtw89 doesn't know how to handle. This one could be "FWCMD_C2H_FUNC_BCN_UPD_DONE". It makes sense to see this only in AP mode.

dubhater, 2025-07-22:
@morrownr I made the message go away.

morrownr, 2025-07-22:
@Kiwimannen 

> I can get things going and mail you when I run into problems - thanks :)

Great! Before you need the example `hostapd-rtl8852cu.conf` file, I am making some updates, so let me know when you need the updated file and I will send it to you.

One last thing:

```
country SE: DFS-ETSI
	(2400 - 2483.5 @ 40), (100 mW)
	(5150 - 5250 @ 80), (200 mW), NO-OUTDOOR, AUTO-BW, wmmrule=ETSI
	(5250 - 5350 @ 80), (100 mW), NO-OUTDOOR, DFS, AUTO-BW, wmmrule=ETSI
	(5470 - 5725 @ 160), (500 mW), DFS, wmmrule=ETSI
	# short range devices (ETSI EN 300 440-1)
	(5725 - 5875 @ 80), (25 mW)
	# 6 GHz band
	(5945 - 6425 @ 320), (23), NO-OUTDOOR, wmmrule=ETSI
	# 60 GHz band channels 1-4 (ETSI EN 302 567)
	(57000 - 66000 @ 2160), (40)
```

Here is what the regdb shows for your country. We need to sort through that to make sure we know what we are doing.

Kiwimannen, 2025-07-22:
I have looked through the hostapd log and found some errors  regarding this part:

```
# 6 GHz Access Point type
2 = Very low power AP (default)

# This has no impact for operation on other bands.
he_6ghz_reg_pwr_type=0
```
The log said unknown option " 2 " (line 119)

and that something was wrong on line 122 : he_6ghz_reg_pwr_type=0

I tried to just comment both out - but  then hostapd says it cant find my wlan adapter.

If you have some alterations I can try here first, I will do that and then start with a new installation tomorrow if it does not work. 

I have several Rasp 4b and many sd-cards so I can try some different configurations without erasing the old ones.

You can send the file with mail whenever you want :)

dubhater, 2025-07-23:
> [@dubhater](https://github.com/dubhater)
> 
> > I made the message go away.
> 
> I'm good with that. Thanks.
> 
> FYI: Update on my AP mode test:
> 
>     1. WPA3 testing. It now appears to be working and I have idea what caused it. It is a mystery.
> 

Haha, just when I decided to try it myself. (It worked for me too.)

>     2. Without 80+80 support, I do not have the capability to test 160 MHz channel width on 5 GHz given my country.
> 
> 
> Overall, AP mode for the rtl8852/32cu is really stable. There are some reported vs actual capability issues I can go into detail at some point. I'm thinking of adding an example `hostapd-rtl8852cu.conf` file to the repo here like I do with the vendor drivers. What do you think?
> 

It's a good idea.

> Therefore, here are my plans:
> 
> Start AP mode testing with the rtl8862bu adapter to see what the status is.
> 
> Work with [@Kiwimannen](https://github.com/Kiwimannen) to get him going with 6 GHz AP mode testing.

So yesterday I tested WPA3 in AP mode. Today I wanted to try 6 GHz AP mode. It doesn't work. It says "Failed to set beacon parameters" no matter what I do. Adding -d reveals the error code is -95 (EOPNOTSUPP). It turns out rtw89 doesn't want to support AP mode in the 6 GHz band:

```C
static int rtw89_ops_start_ap(struct ieee80211_hw *hw,
			      struct ieee80211_vif *vif,
			      struct ieee80211_bss_conf *link_conf)
{
	struct rtw89_dev *rtwdev = hw->priv;
	struct rtw89_vif *rtwvif = vif_to_rtwvif(vif);
	struct rtw89_vif_link *rtwvif_link;
	const struct rtw89_chan *chan;
	int ret = 0;

	lockdep_assert_wiphy(hw->wiphy);

	rtwvif_link = rtwvif->links[link_conf->link_id];
	if (unlikely(!rtwvif_link)) {
		rtw89_err(rtwdev,
			  "%s: rtwvif link (link_id %u) is not active\n",
			  __func__, link_conf->link_id);
		return -ENOLINK;
	}

	chan = rtw89_chan_get(rtwdev, rtwvif_link->chanctx_idx);
	if (chan->band_type == RTW89_BAND_6G)
		return -EOPNOTSUPP;
```

If I remove the last two lines the AP starts. The [commit](https://github.com/pkshih/rtw/commit/f6baa1d3d5703fe65b87b3149265a1c7f564f450) which added this doesn't explain why.

morrownr, 2025-07-23:
@dubhater @Kiwimannen 

>> I'm thinking of adding an example hostapd-rtl8852cu.conf file to the repo here like I do with the vendor drivers. What do you think?

> It's a good idea.

I will attempt to make that happen today.

>
```
 	if (chan->band_type == RTW89_BAND_6G)
		return -EOPNOTSUPP;
```

Hmmm... let me suggest that you create a well documented patch removing those two lines and apply it to this repo so that you and @Kiwimannen can test 6 GHz. Depending on what we find, the patch can be submitted to Ping-Ke.

dubhater, 2025-07-23:
Okay, they are removed.

Kiwimannen, 2025-07-23:
This is interesting: one of my RTL8832CU adapters can be set to AP-mode, the other one can not.

Tested with the new code and exact same setup - just changing the adapter and after reboot, one is in managed-mode and one is in AP-mode. 
```

@raspberrypi:~ $ iw dev
phy#1
    Interface wlx90de8043767f
        ifindex 5
        wdev 0x100000001
        addr 90:de:80:43:76:7f
        type AP
        multicast TXQ:
            qsz-byt    qsz-pkt    flows    drops    marks    overlmt    hashcol    tx-bytes    tx-packets
            0    0    0    0    0    0    0    0        0

[    8.177890] usbcore: registered new interface driver rtw89_8852cu_git
[   11.405265] rtw89_8852cu_git 2-1:1.0: loaded firmware rtw89/rtw8852c_fw-2.bin
[   11.405434] rtw89_8852cu_git 2-1:1.0: git commit 
[   11.407105] rtw89_8852cu_git 2-1:1.0: MAC has already powered on
[   11.429872] rtw89_8852cu_git 2-1:1.0: Firmware version 0.27.129.1 (a71be205), cmd version 0, type 1
[   11.429894] rtw89_8852cu_git 2-1:1.0: Firmware version 0.27.129.1 (a71be205), cmd version 0, type 3
[   11.992252] rtw89_8852cu_git 2-1:1.0: chip rfe_type is 1
[   12.032321] rtw89_8852cu_git 2-1:1.0: Firmware element BB version: 00 28 00 00
[   12.032642] rtw89_8852cu_git 2-1:1.0: Firmware element radio A version: 00 63 00 00
[   12.032824] rtw89_8852cu_git 2-1:1.0: Firmware element NCTL version: 00 11 00 00
[   12.033246] rtw89_8852cu_git 2-1:1.0: Firmware element TXPWR version: 00 78 00 00
[   12.033256] rtw89_8852cu_git 2-1:1.0: Firmware element PWR_TRK version: 00 26 00 00
[   12.033272] rtw89_8852cu_git 2-1:1.0: Firmware element REGD version: 00 48 00 39
[   12.040584] rtw89_8852cu_git 2-1:1.0: rfkill hardware state changed to enable
[   12.071557] rtw89_8852cu_git 2-1:1.0 wlx90de8043767f: renamed from wlan0
```

Checking hostapd.conf after I started with the adapter i AP mode among other things I find:

`the driver does not support the configured HE 6GHz Tx Antenna Pattern`

So Im really interested in that hostapd-rtl8852cu.conf file to test :)

morrownr, 2025-07-24:
Update: I took a flight to Sweden to see what I could see. Not really but my test system thinks it is there and it emailed me that it is a beautiful place.

I have been trying several things. I still cannot get hostapd beyond:

The driver does not support the configured HE 6 GHz Tx Antenna Pattern

I have tried both settings and have even changed some other things that I thought could possibly have an effect on this. No go.

I'll admit that I have very limited experience with 6 GHz in AP mode. I think I am going to have to do some research. There is little prior art for us to use and what little prior art their is for 6 GHz USB WiFi adapters is mostly for Mediatek since Mediatek has had tri-band adapters for several more years than Realtek... and good drivers for more time also. This rtl8852/32cu chip is the first one from Realtek that is triband whereas Mediatek's mt7921 chip goes back a few years. It is not clear to me why Realtek got so far behind Mediatek in technology but they did. Last week I was able to buy a Netgear A9000, which is the first WiFi 7 adapter that is available that has a Linux driver... that has been in the kernel since kernel 6.7.

This problem we are seeing very well could be something in the driver. It was concerning that dubhater had to turn on 6 GHz AP mode... which leads to the question of whether AP mode has ever been finished in the core rtw89 code.

@Kiwimannen 

Remember, with the changes that are going in, you need to remember to run:

$ git pull
then uninstall and install again... to include firmware.

@dubhater 

Any thoughts?

Kiwimannen, 2025-07-24:
Im gonna try to run your guide on Ubuntu tomorrow - if I make a simple "hotspot" with my remaining adapter in Raspberry Pi OS it still shows as managed by iw dev and I cant see the SSID when I scan with another computer.

However, if I do the the same thing in Ubuntu it actually shows as AP with iw dev and I can see the SSID and connect (in 2.4GHz)

morrownr, 2025-07-25:
@Kiwimannen 

Recommend you discontinue work with the current RasPiOS. I just cloned the hostap repo. I had an old copy of the v2.10 source. I did a comparison. No wonder we are struggling with WiFi 6. My bad. Give me a chance to do some compiling, then I will recommend what we should do.

Reminder to self: RasPiOS is based on Debian, which always contains a lot of old stuff and it appears that RasPiOS is not updating hostapd. Arg...

Kiwimannen, 2025-07-25:
> I know going through the hostapd.log is a pain in the ass if you are not familiar with it but that is your best bet to find out why hostapd is not going into AP mode.

No, i dont think its a pain in the ass - but my last two installations (one on raspi and this one on ubuntu) does not give a hostapd.log file. I have checked that my path is correct when setting up with the guide and also searched the whole system after the logfile, but it cant find it - will investigate more why its happening.

dubhater, 2025-07-25:
You can just run hostapd manually, then it prints to standard output/error.

Kiwimannen, 2025-07-25:
> You can just run hostapd manually, then it prints to standard output/error.

Thanks, but I fixed it now - my line was in the wrong section of the file.
Going to look through and see what I can find.

morrownr, 2025-07-28:
Update:

I need to be on the road for a few days for personal reasons but here is an update on where I am with 6 GHz:

I am questioning whether AP mode is finished for 6 GHz in rtw89. Just to get an idea if I was on the right road, I grabbed a tri-band adapter based on the mt7921au chip.

```
$ iw dev
phy#0
	Interface wlan0
		ifindex 4
		wdev 0x1
		addr e8:4e:06:aa:84:9b
		ssid mt7921au
		type AP
		channel 69 (6295 MHz), width: 80 MHz, center1: 6305 MHz
		txpower 3.00 dBm
		multicast TXQ:
			qsz-byt	qsz-pkt	flows	drops	marks	overlmt	hashcol	tx-bytes	tx-packets
			0	0	3342	0	0	0	0	1119432		3501

```

Interestingly, I was able to bring up this chip/driver up in AP mode. I was able to connect with a Win 11 client and things seemed to work well in short testing. However none of my Linux systems could see the SSID. Any ideas?

Of note: my RasPi4B that serves as the AP mode test system had to take a trip due to wireless-regdb issue with 6 GHz in my country:

```
$ iw reg get
global
country SE: DFS-ETSI
	(2400 - 2483 @ 40), (N/A, 20), (N/A)
	(5150 - 5250 @ 80), (N/A, 23), (N/A), NO-OUTDOOR, AUTO-BW
	(5250 - 5350 @ 80), (N/A, 20), (0 ms), NO-OUTDOOR, DFS, AUTO-BW
	(5470 - 5725 @ 160), (N/A, 26), (0 ms), DFS
	(5725 - 5875 @ 80), (N/A, 13), (N/A)
	(5945 - 6425 @ 320), (N/A, 23), (N/A), NO-OUTDOOR
	(57000 - 66000 @ 2160), (N/A, 40), (N/A)

```

I will continue work as able. Let me know if you have success.

Kiwimannen, 2025-07-29:
Sorry for my late answer, but dubhaters hint pointed me in the right direction and noticed a small error in my config - I was sure that was the problem, but after a restart it still does not work for me.

Now I do have some "key" problems in hostapd log that im looking at instead to make some corrections in the config.

My clients (one Wifi 6E Android phone, One Wifi7 Laptop running Ubuntu, One Windows laptop running Wifi7 with the REL8912AU usb-adapter) seems to connect since I can get information such as "connected at 2882 Mb/s (6,1 GHz)"

In my case the Windows laptop running windows 11 is having trouble seeing the RTL8832CU adapter AP SSID

morrownr, 2025-08-01:
@Kiwimannen 

I should be able to start working on this 6 GHz AP mode issue again soon, hopefully tomorrow. I have been pondering the issues we are running into. I think I am going to do some research. Since there is very little prior art, I think I will dig in over at the OpenWRT site.

dubhater, 2025-09-21:
One of the problems I was seeing is fixed now. The other one looks like it won't be fixed in time for 6.18.