morrownr/rtw89 #18: Suspending issue

mrcn49, 2025-07-16:
When i resume from suspending my wifi is gone unfortunately
It's searching for wifi endlessly

l```
ip 16 23:20:18 fedora kernel: printk: Suspending console(s) (use no_console_suspend to debug)
lip 16 23:20:18 fedora kernel: PM: suspend devices took 0.218 seconds
lip 16 23:20:18 fedora kernel: pcieport 0000:00:1c.7: broken device, retraining non-functional downstream link at 2.5GT/s
lip 16 23:20:18 fedora kernel: pcieport 0000:00:1c.7: broken device, retraining non-functional downstream link at 2.5GT/s
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 1023ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 2047ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 4095ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 8191ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 16383ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 32767ms after resume; waiting
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: not ready 65535ms after resume; giving up
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: Unable to change power state from D3cold to D0, device inaccessible
lip 16 23:20:18 fedora kernel: i915 0000:00:02.0: [drm] GT0: GuC firmware i915/adlp_guc_70.bin version 70.44.1
lip 16 23:20:18 fedora kernel: i915 0000:00:02.0: [drm] GT0: HuC firmware i915/tgl_huc.bin version 7.9.3
lip 16 23:20:18 fedora kernel: spd5118 16-0050: PM: dpm_run_callback(): spd5118_resume [spd5118] returns -6
lip 16 23:20:18 fedora kernel: spd5118 16-0050: PM: failed to resume async: error -6
lip 16 23:20:18 fedora kernel: spd5118 16-0052: PM: dpm_run_callback(): spd5118_resume [spd5118] returns -6
lip 16 23:20:18 fedora kernel: spd5118 16-0052: PM: failed to resume async: error -6
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: failed to write DBI register, addr=0xB48
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: failed to write DBI register, addr=0xB48
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: failed to read PCI cap, ret=134
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: xtal si not ready(W): offset=90 val=10 mask=10
lip 16 23:20:18 fedora kernel: rtw89_8852be_git 0000:39:00.0: mac init fail, ret:-110
lip 16 23:20:18 fedora kernel: Hardware became unavailable upon resume. This could be a software issue prior to suspend or a hardware issue.
lip 16 23:20:18 fedora kernel: WARNING: CPU: 4 PID: 4501 at net/mac80211/util.c:1829 ieee80211_reconfig+0x99/0x12b0 [mac80211]
lip 16 23:20:18 fedora kernel: Modules linked in: overlay uinput rfcomm snd_seq_dummy snd_hrtimer nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject sunrpc nft_ct nft_chain_nat nf_nat nf_conntrack snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel nf_defrag_ipv6 snd_sof_intel_hda_sdw_bpt nf_defrag_ipv4 snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda soundwire_cadence snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi crc8 soundwire_bus snd_soc_sdca snd_soc_avs snd_soc_hda_codec snd_hda_ext_core snd_soc_core nf_tables snd_compress ac97_bus snd_pcm_dmaengine qrtr bnep snd_hda_codec_hdmi vfat fat snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component intel_uncore_frequency intel_uncore_frequency_common intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp
lip 16 23:20:18 fedora kernel:  coretemp rtw89_8852be_git(OE) rtw89_8852b_git(OE) snd_hda_intel rtw89_8852b_common_git(OE) rtw89_pci_git(OE) kvm_intel snd_intel_dspcfg nvidia_drm(OE) rtw89_core_git(OE) snd_intel_sdw_acpi nvidia_modeset(OE) snd_hda_codec btusb mac80211 kvm snd_hda_core btrtl btintel snd_hwdep nvidia_uvm(OE) btbcm processor_thermal_device_pci uvcvideo spi_nor libarc4 snd_seq processor_thermal_device btmtk irqbypass processor_thermal_wt_hint uvc mtd iTCO_wdt snd_seq_device asus_nb_wmi processor_thermal_rfim intel_pmc_bxt rapl spd5118 mei_hdcp mei_pxp iTCO_vendor_support intel_rapl_msr videobuf2_vmalloc cfg80211 bluetooth snd_pcm asus_wmi intel_cstate processor_thermal_rapl spi_intel_pci videobuf2_memops sparse_keymap r8169 i2c_i801 snd_timer videobuf2_v4l2 nvidia(OE) platform_profile intel_rapl_common mei_me wmi_bmof intel_uncore videobuf2_common pcspkr nvidia_wmi_ec_backlight thunderbolt spi_intel rfkill snd i2c_smbus processor_thermal_wt_req realtek ov13858 processor_thermal_power_floor mei v4l2_fwnode idma64 soundcore
lip 16 23:20:18 fedora kernel:  processor_thermal_mbox igen6_edac v4l2_async videodev int3403_thermal mc int340x_thermal_zone intel_pmc_core int3400_thermal pmt_telemetry acpi_pad acpi_thermal_rel pmt_class acpi_tad joydev loop nfnetlink zram lz4hc_compress lz4_compress xe intel_vsec drm_ttm_helper drm_suballoc_helper gpu_sched drm_gpuvm drm_exec drm_gpusvm i915 i2c_algo_bit drm_buddy ttm nvme nvme_core drm_display_helper polyval_clmulni polyval_generic ucsi_acpi ghash_clmulni_intel sha512_ssse3 typec_ucsi sha256_ssse3 hid_multitouch nvme_keyring cec sha1_ssse3 typec nvme_auth video i2c_hid_acpi wmi i2c_hid pinctrl_tigerlake serio_raw i2c_dev fuse
lip 16 23:20:18 fedora kernel:  wiphy_resume+0x82/0x1b0 [cfg80211]
lip 16 23:20:18 fedora kernel:  ? __pfx_wiphy_resume+0x10/0x10 [cfg80211]
lip 16 23:20:18 fedora kernel:  device_resume+0x152/0x250
lip 16 23:20:18 fedora kernel:  async_resume+0x21/0x30
lip 16 23:20:18 fedora kernel: WARNING: CPU: 4 PID: 4501 at net/mac80211/driver-ops.c:41 drv_stop+0xfd/0x110 [mac80211]
lip 16 23:20:18 fedora kernel: Modules linked in: overlay uinput rfcomm snd_seq_dummy snd_hrtimer nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 nft_reject sunrpc nft_ct nft_chain_nat nf_nat nf_conntrack snd_sof_pci_intel_tgl snd_sof_pci_intel_cnl snd_sof_intel_hda_generic soundwire_intel nf_defrag_ipv6 snd_sof_intel_hda_sdw_bpt nf_defrag_ipv4 snd_sof_intel_hda_common snd_soc_hdac_hda snd_sof_intel_hda_mlink snd_sof_intel_hda soundwire_cadence snd_sof_pci snd_sof_xtensa_dsp snd_sof snd_sof_utils snd_soc_acpi_intel_match snd_soc_acpi_intel_sdca_quirks soundwire_generic_allocation snd_soc_acpi crc8 soundwire_bus snd_soc_sdca snd_soc_avs snd_soc_hda_codec snd_hda_ext_core snd_soc_core nf_tables snd_compress ac97_bus snd_pcm_dmaengine qrtr bnep snd_hda_codec_hdmi vfat fat snd_hda_codec_realtek snd_hda_codec_generic snd_hda_scodec_component intel_uncore_frequency intel_uncore_frequency_common intel_tcc_cooling x86_pkg_temp_thermal intel_powerclamp
lip 16 23:20:18 fedora kernel:  coretemp rtw89_8852be_git(OE) rtw89_8852b_git(OE) snd_hda_intel rtw89_8852b_common_git(OE) rtw89_pci_git(OE) kvm_intel snd_intel_dspcfg nvidia_drm(OE) rtw89_core_git(OE) snd_intel_sdw_acpi nvidia_modeset(OE) snd_hda_codec btusb mac80211 kvm snd_hda_core btrtl btintel snd_hwdep nvidia_uvm(OE) btbcm processor_thermal_device_pci uvcvideo spi_nor libarc4 snd_seq processor_thermal_device btmtk irqbypass processor_thermal_wt_hint uvc mtd iTCO_wdt snd_seq_device asus_nb_wmi processor_thermal_rfim intel_pmc_bxt rapl spd5118 mei_hdcp mei_pxp iTCO_vendor_support intel_rapl_msr videobuf2_vmalloc cfg80211 bluetooth snd_pcm asus_wmi intel_cstate processor_thermal_rapl spi_intel_pci videobuf2_memops sparse_keymap r8169 i2c_i801 snd_timer videobuf2_v4l2 nvidia(OE) platform_profile intel_rapl_common mei_me wmi_bmof intel_uncore videobuf2_common pcspkr nvidia_wmi_ec_backlight thunderbolt spi_intel rfkill snd i2c_smbus processor_thermal_wt_req realtek ov13858 processor_thermal_power_floor mei v4l2_fwnode idma64 soundcore
lip 16 23:20:18 fedora kernel:  processor_thermal_mbox igen6_edac v4l2_async videodev int3403_thermal mc int340x_thermal_zone intel_pmc_core int3400_thermal pmt_telemetry acpi_pad acpi_thermal_rel pmt_class acpi_tad joydev loop nfnetlink zram lz4hc_compress lz4_compress xe intel_vsec drm_ttm_helper drm_suballoc_helper gpu_sched drm_gpuvm drm_exec drm_gpusvm i915 i2c_algo_bit drm_buddy ttm nvme nvme_core drm_display_helper polyval_clmulni polyval_generic ucsi_acpi ghash_clmulni_intel sha512_ssse3 typec_ucsi sha256_ssse3 hid_multitouch nvme_keyring  [... cut ...]
lip 16 23:21:10 fedora NetworkManager[1297]: <error> [1752700870.5942] device (wlp57s0): Couldn't initialize supplicant interface: GDBus.Error:fi.w1.wpa_supplicant1.UnknownError: wpa_supplicant couldn't grab this interface.
lip 16 23:21:10 fedora NetworkManager[1297]: <info>  [1752700870.5942] device (wlp57s0): supplicant interface keeps failing, giving up
lip 16 23:21:10 fedora kernel: rtw89_8852be_git 0000:39:00.0: mac init fail, ret:-110
lip 16 23:22:03 fedora sudo[4995]:   marcin : TTY=pts/0 ; PWD=/home/marcin ; USER=root ; COMMAND=/usr/sbin/find /lib/modules/6.15.5-200.fc42.x86_64/ -name rtw89*
marcin@fedora:~$ 

```

mrcn49, 2025-07-17:
nvm writing 0 to /sys/bus/pci/devices/0000:39:00.0/d3cold_allowed solved my problem

dubhater, 2025-07-17:
You should still report this to linux-wireless@vger.kernel.org. Maybe they can fix it so you don't have to write anything to /sys.