morrownr/rtw89 #20: Throughput with rtw89_8852bu as AP is 2-3x lower as with rtl8852bu

bauner, 2025-07-19:
Following my previous 8852bu AP data throughput tests (https://github.com/morrownr/rtw89/issues/15#issuecomment-3084598565) with a smartphone, where the rtw89 driver was approximately 6x slower, I now repeated the tests with an 8852be (Linux Kernel 6.15 rtw89 driver) as an iperf3 client and a D-Link AX18U (8852bu) as an iperf3 server and access point with the rtl8852bu and rtw89 drivers from https://github.com/morrownr.

**The rtw89 driver is also 2-3 times slower in data throughput than the rtl8852bu driver in these tests.**

**Here are the iperf3 measurements performed on the (8852be) client PC:**
Both drivers in USB 3 mode
```
    |__ Port 004: Dev 002, If 0, Class=Vendor Specific Class, Driver=rtl8852bu, 5000M ID 2001:3327 D-Link Corp.
   |__ Port 004: Dev 002, If 0, Class=Vendor Specific Class, Driver=rtw89_8852bu_git, 5000M ID 2001:3327 D-Link Corp.
```

**iperf3 -c 192.168.178.95 (AP with rtl8852bu driver):**
```
Connecting to host 192.168.178.95, port 5201
[  5] local 192.168.178.101 port 43938 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  64.4 MBytes   539 Mbits/sec    0   2.35 MBytes       
[  5]   1.00-2.00   sec  68.8 MBytes   577 Mbits/sec    0   3.27 MBytes       
[  5]   2.00-3.00   sec  88.8 MBytes   744 Mbits/sec    0   3.27 MBytes       
[  5]   3.00-4.00   sec  87.0 MBytes   730 Mbits/sec    0   3.27 MBytes       
[  5]   4.00-5.00   sec  87.9 MBytes   737 Mbits/sec    0   3.27 MBytes       
[  5]   5.00-6.00   sec  88.5 MBytes   742 Mbits/sec    0   3.27 MBytes       
[  5]   6.00-7.00   sec  87.5 MBytes   734 Mbits/sec    0   3.27 MBytes       
[  5]   7.00-8.00   sec  85.5 MBytes   717 Mbits/sec    0   3.27 MBytes       
[  5]   8.00-9.00   sec  90.1 MBytes   756 Mbits/sec    0   3.27 MBytes       
[  5]   9.00-10.00  sec  87.5 MBytes   734 Mbits/sec    0   3.27 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   836 MBytes   701 Mbits/sec    0            sender
[  5]   0.00-10.01  sec   834 MBytes   699 Mbits/sec                  receiver

iperf Done.
```

**iperf3 -c 192.168.178.95 (AP with rtw89_8852bu_git driver):**
```
Connecting to host 192.168.178.95, port 5201
[  5] local 192.168.178.101 port 42338 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  36.6 MBytes   307 Mbits/sec    0   1.35 MBytes       
[  5]   1.00-2.00   sec  38.4 MBytes   322 Mbits/sec    0   1.71 MBytes       
[  5]   2.00-3.00   sec  30.6 MBytes   257 Mbits/sec    0   1.80 MBytes       
[  5]   3.00-4.00   sec  27.5 MBytes   231 Mbits/sec    0   1.90 MBytes       
[  5]   4.00-5.00   sec  29.8 MBytes   250 Mbits/sec    0   2.01 MBytes       
[  5]   5.00-6.00   sec  28.2 MBytes   237 Mbits/sec    0   2.13 MBytes       
[  5]   6.00-7.00   sec  30.6 MBytes   257 Mbits/sec    0   2.24 MBytes       
[  5]   7.00-8.00   sec  17.4 MBytes   146 Mbits/sec    0   2.24 MBytes       
[  5]   8.00-9.00   sec  17.4 MBytes   146 Mbits/sec    0   2.24 MBytes       
[  5]   9.00-10.00  sec  17.2 MBytes   145 Mbits/sec    0   2.24 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   274 MBytes   230 Mbits/sec    0            sender
[  5]   0.00-10.02  sec   272 MBytes   227 Mbits/sec                  receiver

iperf Done.

```
**iperf3 -c 192.168.178.95 -R (AP with rtl8852bu driver):**
```
Connecting to host 192.168.178.95, port 5201
Reverse mode, remote host 192.168.178.95 is sending
[  5] local 192.168.178.101 port 52930 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  7.88 MBytes  66.0 Mbits/sec                  
[  5]   1.00-2.00   sec  15.2 MBytes   128 Mbits/sec                  
[  5]   2.00-3.00   sec  44.8 MBytes   375 Mbits/sec                  
[  5]   3.00-4.00   sec  44.9 MBytes   376 Mbits/sec                  
[  5]   4.00-5.00   sec  47.9 MBytes   402 Mbits/sec                  
[  5]   5.00-6.00   sec  51.2 MBytes   430 Mbits/sec                  
[  5]   6.00-7.00   sec  53.2 MBytes   447 Mbits/sec                  
[  5]   7.00-8.00   sec  54.8 MBytes   459 Mbits/sec                  
[  5]   8.00-9.00   sec  54.9 MBytes   460 Mbits/sec                  
[  5]   9.00-10.00  sec  56.1 MBytes   471 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.01  sec   434 MBytes   364 Mbits/sec    4            sender
[  5]   0.00-10.00  sec   431 MBytes   361 Mbits/sec                  receiver

iperf Done.

```
**iperf3 -c 192.168.178.95 -R (AP with rtw89_8852bu_git driver):**
```
Connecting to host 192.168.178.95, port 5201
Reverse mode, remote host 192.168.178.95 is sending
[  5] local 192.168.178.101 port 57000 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate
[  5]   0.00-1.00   sec  16.2 MBytes   136 Mbits/sec                  
[  5]   1.00-2.00   sec  24.0 MBytes   201 Mbits/sec                  
[  5]   2.00-3.00   sec  21.5 MBytes   180 Mbits/sec                  
[  5]   3.00-4.00   sec  21.0 MBytes   176 Mbits/sec                  
[  5]   4.00-5.00   sec  25.5 MBytes   214 Mbits/sec                  
[  5]   5.00-6.00   sec  25.1 MBytes   211 Mbits/sec                  
[  5]   6.00-7.00   sec  25.1 MBytes   211 Mbits/sec                  
[  5]   7.00-8.00   sec  25.0 MBytes   210 Mbits/sec                  
[  5]   8.00-9.00   sec  25.1 MBytes   211 Mbits/sec                  
[  5]   9.00-10.00  sec  22.8 MBytes   191 Mbits/sec                  
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   235 MBytes   197 Mbits/sec    0            sender
[  5]   0.00-10.00  sec   231 MBytes   194 Mbits/sec                  receiver

iperf Done.
```

**iperf3 -c 192.168.178.95 -bidir (AP with rtl8852bu driver):**
```
Connecting to host 192.168.178.95, port 5201
[  5] local 192.168.178.101 port 56242 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  64.2 MBytes   539 Mbits/sec    0   2.43 MBytes       
[  5]   1.00-2.00   sec  75.0 MBytes   629 Mbits/sec    0   3.29 MBytes       
[  5]   2.00-3.00   sec  86.1 MBytes   723 Mbits/sec    0   3.46 MBytes       
[  5]   3.00-4.00   sec  88.9 MBytes   746 Mbits/sec    0   3.90 MBytes       
[  5]   4.00-5.00   sec  89.5 MBytes   751 Mbits/sec    0   4.13 MBytes       
[  5]   5.00-6.00   sec  88.4 MBytes   742 Mbits/sec    0   4.13 MBytes       
[  5]   6.00-7.00   sec  88.0 MBytes   738 Mbits/sec    0   4.13 MBytes       
[  5]   7.00-8.00   sec  85.8 MBytes   719 Mbits/sec    0   4.13 MBytes       
[  5]   8.00-9.00   sec  88.9 MBytes   746 Mbits/sec    0   4.13 MBytes       
[  5]   9.00-10.00  sec  88.8 MBytes   744 Mbits/sec    0   4.13 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   844 MBytes   707 Mbits/sec    0            sender
[  5]   0.00-10.01  sec   841 MBytes   704 Mbits/sec                  receiver

iperf Done.
```

**iperf3 -c 192.168.178.95 -bidir (AP with rtw89_8852bu_git driver):**
```
Connecting to host 192.168.178.95, port 5201
[  5] local 192.168.178.101 port 50290 connected to 192.168.178.95 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec  39.4 MBytes   330 Mbits/sec    0   1.95 MBytes       
[  5]   1.00-2.00   sec  41.2 MBytes   346 Mbits/sec    0   3.72 MBytes       
[  5]   2.00-3.00   sec  46.4 MBytes   389 Mbits/sec    0   3.92 MBytes       
[  5]   3.00-4.00   sec  42.5 MBytes   357 Mbits/sec    0   3.92 MBytes       
[  5]   4.00-5.00   sec  42.8 MBytes   359 Mbits/sec    0   3.92 MBytes       
[  5]   5.00-6.00   sec  43.0 MBytes   361 Mbits/sec    0   3.92 MBytes       
[  5]   6.00-7.00   sec  44.6 MBytes   374 Mbits/sec    0   3.92 MBytes       
[  5]   7.00-8.00   sec  43.0 MBytes   361 Mbits/sec    0   3.92 MBytes       
[  5]   8.00-9.00   sec  41.0 MBytes   344 Mbits/sec    0   3.92 MBytes       
[  5]   9.00-10.00  sec  42.9 MBytes   359 Mbits/sec    0   3.92 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec   427 MBytes   358 Mbits/sec    0            sender
[  5]   0.00-10.02  sec   426 MBytes   356 Mbits/sec                  receiver

iperf Done.

```
**Here is the used hostapd.conf:**
https://pastebin.com/NkGpyVVK

In the next days I will test the 8852bu adapter in client mode with the 8852be as an access point.

morrownr, 2025-07-19:
Hi @bauner 

Thanks for the report. I am currently testing AP mode with the rtl8852cu driver in this repo.

Try this change with vht_capab:

[MAX-MPDU-11454]

As I have time I will also test with my rtl8832bu based adapter.

bauner, 2025-07-19:
rtl8852bu with vht_capab [MAX-MPDU-11454] is not possible
**Configured VHT capability [VHT_CAP_MAX_MPDU_LENGTH_MASK] exceeds max value supported by the driver (2 > 1)**

rtw89_8852bu_git with vht_capab [MAX-MPDU-11454] is working but the throughput is nearly the same as in my first post in this thread

morrownr, 2025-07-19:
@bauner 

> rtl8852bu with vht_capab [MAX-MPDU-11454] is not possible

I am aware of that. I am running into several differences between rtl8852cu (vendor driver) and the driver in this repo for the rtl8852cu chip. I assume some of the differences will be in the drivers for the rtl8852bu as well.

I am seeing less of a performance differential than you are. It is going to take some time to sort everything out.

a5a5aa555oo, 2025-07-22:
I think the slower speed is normal because it seems that TX/RX aggregation are not implemented yet.