morrownr/rtw89 #21: Mpdel TXEU70UH does not work for me

martinellison, 2025-07-26:
I have bought a TXEU70UH and it is not working for me (on 6.15.7). Fedora does not display Wi-Fi at all in the control panel.

The model 50 seems not to be on sale in Sham Shui Po. I presume that it is obsolete now.

https://github.com/morrownr has advised me to raise this here.

a5a5aa555oo, 2025-07-26:
TXE**U**70UH or TXE70UH? typo?

Please post the output of the `lsusb` command.

There are 6 IDs in TXE70UH's Windows driver, we need to know what ID your Wi-Fi dongle has.

[image]

morrownr, 2025-07-26:
@martinellison 

> The model 50 seems not to be on sale in Sham Shui Po. I presume that it is obsolete now.

The TXE50UH is a recent release. It is a WiFi 6 tri-band adapter. It could be that your retailer does not carry it yet.

martinellison, 2025-08-04:
>
> ~> lsusb
> Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 001 Device 002: ID 2a94:736d G2Touch Multi-Touch by G2TSP
> Bus 001 Device 004: ID 35bc:0102 Realtek 802.11ax WLAN Adapter
> Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 002 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub
> Bus 002 Device 003: ID 0781:cff8 SanDisk Corp. iXpand Flash Drive
> Bus 002 Device 004: ID 1532:006e Razer USA, Ltd DeathAdder Essential
> Bus 002 Device 005: ID 1c4f:0002 SiGma Micro Keyboard TRACER Gamma Ivory
> Bus 003 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
> Bus 004 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
> Bus 004 Device 002: ID 8087:0024 Intel Corp. Integrated Rate Matching Hub

Sorry for the delay; have been busy.

The device calls itself TXE70UH US Ver:1.0 [the 0 could be an O].

morrownr, 2025-08-04:
@martinellison 

>  Bus 001 Device 004: ID 35bc:0102 Realtek 802.11ax WLAN Adapter

The above is your adapter. I checked in rtw8852cu.c and found the following:

```
	{ USB_DEVICE_AND_INTERFACE_INFO(0x35bc, 0x0102, 0xff, 0xff, 0xff),
	  .driver_info = (kernel_ulong_t)&rtw89_8852cu_info },
```

Your adapter's device ID is in the driver so that is not the problem.

> I have bought a TXEU70UH and it is not working for me (on 6.15.7). Fedora does not display Wi-Fi at all in the control panel.

Did you install the driver in this repo?

Did you uninstall any previous drivers that you may have installed?