morrownr/rtw89 #27: RTW8922au USB Test Results

it-is-i-dhruv, 2025-09-02:
Hello, am using the following system:

Adaptor: TP-Link TBE400UH

OS: 6.16.4-arch1-1

Laptop: HP Victus

And seem to not be able to use the adaptor after the first few packets, after which I see this error on dmesg:

```
14.268266] wlan0: authenticate with 80:76:b9:84:78:ef (local address=48:e7:da:cc:ab:71)
[   14.268271] wlan0: send auth to 80:76:b9:84:78:ef (try 1/3)
[   14.278753] wlan0: authenticated
[   14.279340] wlan0: associate with 80:76:b9:84:78:ef (try 1/3)
[   14.281058] wlan0: RX AssocResp from 80:76:b9:84:78:ef (capab=0x1931 status=0 aid=1)
[   14.391466] wlan0: associated
[   14.391492] wlan0: Limiting TX power to 30 (30 - 0) dBm as advertised by 80:76:b9:84:78:ef
[   15.372527] rtw89_8922au_git 4-1:1.0: failed to wait RF TSSI
[   15.407530] wlan1: authenticate with 80:76:b9:84:78:ef (local address=e0:d3:62:5f:1b:06)
[   15.407537] wlan1: send auth to 80:76:b9:84:78:ef (try 1/3)
[   15.418012] wlan1: authenticated
[   15.418331] wlan1: associate with 80:76:b9:84:78:ef (try 1/3)
[   15.420169] wlan1: RX AssocResp from 80:76:b9:84:78:ef (capab=0x1931 status=0 aid=2)
[   15.530965] wlan1: associated
[   15.530994] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 80:76:b9:84:78:ef
[   20.799676] warning: `ThreadPoolForeg' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[   20.994435] Bluetooth: RFCOMM TTY layer initialized
[   20.994452] Bluetooth: RFCOMM socket layer initialized
[   20.994458] Bluetooth: RFCOMM ver 1.11
[   31.743905] wlan1: Connection to AP 80:76:b9:84:78:ef lost
[   31.974434] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   32.215393] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   32.429596] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   36.616358] rtw89_8922au_git 4-1:1.0: failed to wait RF TSSI
[   36.643224] wlan1: authenticate with 80:76:b9:84:78:ef (local address=e0:d3:62:5f:1b:06)
[   36.643227] wlan1: send auth to 80:76:b9:84:78:ef (try 1/3)
[   36.686707] wlan1: authenticated
[   36.688336] wlan1: associate with 80:76:b9:84:78:ef (try 1/3)
[   36.690523] wlan1: RX AssocResp from 80:76:b9:84:78:ef (capab=0x1931 status=0 aid=2)
[   36.801954] wlan1: associated
[   36.801990] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 80:76:b9:84:78:ef
[   49.121510] wlan0: deauthenticating from 80:76:b9:84:78:ef by local choice (Reason: 3=DEAUTH_LEAVING)
[   65.776842] wlan1: Connection to AP 80:76:b9:84:78:ef lost
[   66.000467] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   66.277407] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   66.489441] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   71.224330] rtw89_8922au_git 4-1:1.0: failed to wait RF TSSI
[   71.251070] wlan1: authenticate with 80:76:b9:84:78:ef (local address=e0:d3:62:5f:1b:06)
[   71.251074] wlan1: send auth to 80:76:b9:84:78:ef (try 1/3)
[   71.297163] wlan1: authenticated
[   71.298331] wlan1: associate with 80:76:b9:84:78:ef (try 1/3)
[   71.299933] wlan1: RX AssocResp from 80:76:b9:84:78:ef (capab=0x1931 status=0 aid=1)
[   71.409980] wlan1: associated
[   71.410028] wlan1: Limiting TX power to 30 (30 - 0) dBm as advertised by 80:76:b9:84:78:ef
[   77.645922] wlan1: Connection to AP 80:76:b9:84:78:ef lost
[   77.879465] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   78.132271] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   78.338391] rtw89_8922au_git 4-1:1.0: timed out to flush queues
[   83.358327] rtw89_8922au_git 4-1:1.0: failed to wait RF TSSI
```

`lsusb` outputs:
`Bus 004 Device 002: ID 3625:010a Realtek 802.11be WLAN Adapter`

`journalctl -b -1 | grep Network` outputs:
```
Sep 03 02:28:11 zoltop NetworkManager[950]: <info>  [1756837691.4105] device (p2p-dev-wlan1): supplicant management interface state: interface_disabled -> inactive
Sep 03 02:28:12 zoltop NetworkManager[950]: <info>  [1756837692.0738] device (wlan1): supplicant interface state: inactive -> authenticating
Sep 03 02:28:12 zoltop NetworkManager[950]: <info>  [1756837692.0738] device (p2p-dev-wlan1): supplicant management interface state: inactive -> authenticating
[... 88 lines cut ...]
Sep 03 02:28:43 zoltop NetworkManager[950]: <info>  [1756837723.6987] device (wlan1): supplicant interface state: completed -> disconnected
Sep 03 02:28:43 zoltop NetworkManager[950]: <info>  [1756837723.6988] device (p2p-dev-wlan1): supplicant management interface state: completed -> disconnected
Sep 03 02:28:44 zoltop NetworkManager[950]: <info>  [1756837724.2901] device (wlan1): supplicant interface state: disconnected -> scanning
Sep 03 02:28:44 zoltop NetworkManager[950]: <info>  [1756837724.2901] device (p2p-dev-wlan1): supplicant management interface state: disconnected -> scanning
Sep 03 02:28:46 zoltop systemd[1]: NetworkManager-dispatcher.service: Deactivated successfully.
Sep 03 02:28:47 zoltop NetworkManager[950]: <info>  [1756837727.9574] device (wlan1): supplicant interface state: scanning -> authenticating
Sep 03 02:28:47 zoltop NetworkManager[950]: <info>  [1756837727.9574] device (p2p-dev-wlan1): supplicant management interface state: scanning -> authenticating
Sep 03 02:28:47 zoltop NetworkManager[950]: <info>  [1756837727.9671] device (wlan1): supplicant interface state: authenticating -> associating
Sep 03 02:28:47 zoltop NetworkManager[950]: <info>  [1756837727.9672] device (p2p-dev-wlan1): supplicant management interface state: authenticating -> associating
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.0849] device (wlan1): supplicant interface state: associating -> 4way_handshake
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.0850] device (p2p-dev-wlan1): supplicant management interface state: associating -> 4way_handshake
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1148] device (wlan1): supplicant interface state: 4way_handshake -> completed
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1149] device (wlan1): ip:dhcp4: restarting
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1317] dhcp4 (wlan1): canceled DHCP transaction
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1317] dhcp4 (wlan1): state changed no lease
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1318] dhcp4 (wlan1): activation: beginning transaction (timeout in 45 seconds)
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1319] device (p2p-dev-wlan1): supplicant management interface state: 4way_handshake -> completed
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1742] dhcp4 (wlan1): state changed new lease, address=192.168.0.21, acd pending
Sep 03 02:28:48 zoltop NetworkManager[950]: <info>  [1756837728.1746] dhcp4 (wlan1): state changed new lease, address=192.168.0.21
Sep 03 02:28:48 zoltop systemd[1]: Starting Network Manager Script Dispatcher Service...
```

Not sure if its a dhcp issue or usb issue at this point. The wifi card on the laptop seems to work fine with the dhcp server however.

Thanks

dubhater, 2025-09-02:
Please attach the full output of ``journalctl -b``.

it-is-i-dhruv, 2025-09-03:
Thanks for the suggestions. Here is a new pastebin with `journalctl -b` in it, without the noise from Tailscale and the other driver: https://pastebin.com/2J1J8gbL

I live in Singapore, and the adaptor is from Japan. Wireless-regdb doesn't seem to have helped in the issue. Not sure what is going on either.

dubhater, 2025-09-26:
Yes, it's 160 MHz. I don't think they have a chip capable of 320. At least not for client devices.

morrownr, 2025-09-27:
@dubhater 

> Yes, it's 160 MHz. I don't think they have a chip capable of 320. At least not for client devices.

Thanks. FYI: I'm not what the story behind it is but Mediatek has release a mt7927 chip for client use that support 320. Unfortunately no driver has been released and AMD is using the chip on desktop motherboards and in laptops. I think you knew this.

More testing below:

```
$ ping 192.168.1.1
PING 192.168.1.1 (192.168.1.1) 56(84) bytes of data.
64 bytes from 192.168.1.1: icmp_seq=1 ttl=64 time=0.819 ms
64 bytes from 192.168.1.1: icmp_seq=2 ttl=64 time=0.714 ms
64 bytes from 192.168.1.1: icmp_seq=3 ttl=64 time=0.701 ms
64 bytes from 192.168.1.1: icmp_seq=4 ttl=64 time=0.715 ms
64 bytes from 192.168.1.1: icmp_seq=5 ttl=64 time=0.719 ms
64 bytes from 192.168.1.1: icmp_seq=6 ttl=64 time=0.941 ms
64 bytes from 192.168.1.1: icmp_seq=7 ttl=64 time=0.627 ms
64 bytes from 192.168.1.1: icmp_seq=8 ttl=64 time=0.707 ms
64 bytes from 192.168.1.1: icmp_seq=9 ttl=64 time=0.755 ms
64 bytes from 192.168.1.1: icmp_seq=10 ttl=64 time=0.665 ms
64 bytes from 192.168.1.1: icmp_seq=11 ttl=64 time=0.715 ms
64 bytes from 192.168.1.1: icmp_seq=12 ttl=64 time=0.710 ms
^C
--- 192.168.1.1 ping statistics ---
12 packets transmitted, 12 received, 0% packet loss, time 11210ms
rtt min/avg/max/mdev = 0.627/0.732/0.941/0.076 ms

```

Very low, stable ping times. Impressive.

In over a day of testing, I have seen no drops.

```
$ sudo iw dev
[sudo] password for morrow: 
phy#2
        Interface wlx90de80c702fa
                ifindex 4
                wdev 0x200000001
                addr 90:de:80:c7:02:fa
                ssid Bass
                type managed
                channel 100 (5500 MHz), width: 160 MHz, center1: 5570 MHz
                txpower 20.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0
phy#1
        Interface wlp3s0
                ifindex 3
                wdev 0x100000001
                addr 46:d2:4b:ac:c6:7c
                type managed
                txpower 3.00 dBm
                multicast TXQ:
                        qsz-byt qsz-pkt flows   drops   marks   overlmt hashcol tx-bytes        tx-packets
                        0       0       0       0       0       0       0       0               0

phy#2 is the rtl8912au device.

morrow@vostro:~$ wavemon -i wlx90de80c702fa
wavemon: wlx90de80c702fa is not a usable wireless interface

Never seen that before and have no idea what would cause it. It is obviously unable as my system is connected to the network and is working fine. Unable to use wavemon.

```

Here is a retest of iperf3:
[... 42 lines cut ...]
[   13.329688] rtw89_8922au_git 4-3:1.0 wlx90de80c702fa: renamed from wlan0
[   13.545503] r8169 0000:04:00.0 enp4s0: Link is Up - 1Gbps/Full - flow control rx/tx
[   13.776204] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[   19.200190] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[   27.809962] warning: `kdeconnectd' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211
[  171.289538] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[  171.421566] rtw89_8922au_git 4-3:1.0: failed to wait RF RX_DCK
[  171.433580] rtw89_8922au_git 4-3:1.0: failed to wait RF PRE_NTFY
[  171.497581] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[  171.633622] rtw89_8922au_git 4-3:1.0: failed to wait RF TSSI
[  175.544318] wlx90de80c702fa: authenticate with 82:af:ca:06:da:fd (local address=90:de:80:c7:02:fa)
[  175.544329] wlx90de80c702fa: send auth to 82:af:ca:06:da:fd (try 1/3)
[  175.761952] wlx90de80c702fa: authenticate with 82:af:ca:06:da:fd (local address=90:de:80:c7:02:fa)
[  175.761964] wlx90de80c702fa: send auth to 82:af:ca:06:da:fd (try 1/3)
[  175.768908] wlx90de80c702fa: authenticated
[  175.770603] wlx90de80c702fa: associate with 82:af:ca:06:da:fd (try 1/3)
[  175.774111] wlx90de80c702fa: RX AssocResp from 82:af:ca:06:da:fd (capab=0x111 status=0 aid=2)
[  175.889596] wlx90de80c702fa: associated
[  175.889695] wlx90de80c702fa: Limiting TX power to 21 (24 - 3) dBm as advertised by 82:af:ca:06:da:fd

```

The following line leaves me pondering exactly what it will affect:

warning: `kdeconnectd' uses wireless extensions which will stop working for Wi-Fi 7 hardware; use nl80211

dubhater, 2025-09-28:
wavemon works for me, including the -i switch. But my USB devices are named after their location in the system (wlp3s0f3u4), not the MAC address.

The unreliable performance could be related to those random "failed to wait RF" messages. It looks like the RF calibration is not always successful.

KDE Connect is something you use to communicate with your phone. If you didn't install the corresponding app on your phone then you can ignore that message.

morrownr, 2025-09-29:
@it-is-i-dhruv 

> has the driver had issues with Wifi 7 routers?

I can't test WiFi 7 with a router as I do not have a WiFi 7 capable router yet. I am waiting to see if OpenWRT Two turns out to be a good router.  My testing is on a dual band WiFi 6 router while using channel 100, 160 MHz channel width. You can see that I m getting up to 1.4 Gbps but the speed varies a lot and it should not. Channel 100 is a DFS channel in my country and I am the only one using the DFS channels so it is clear air. The iperf3 output should be rock solid stable at max speed but it is not. The mt7925u based adapter that I have is WiFi 7, gets >1 Gbps in its location and does not vary it its speed.

> slight stress testing of the adaptor gives a crash of the whole driver this time.

I will do some extended time stress tests with iperf3. Should I start with 5 minutes?

> Also isolated to see if its a signal strength issue, but sitting right beside the router does not seem to help either.

Probably not a signal strength issue. Do you run `wavemon`? You can check. It would be interesting.

What band and channel are you using with WiFi 7?

Have you made sure the firmware of your router is up to date?

morrownr, 2025-10-04:
@dubhater 

> -u is limited by default to 1 Mbps. You can add -b 0 to remove the limit.

Got it. Thanks. 

I am continuing to use the driver. I am not seeing the problems that @it-is-i-dhruv reported.

Generally speaking, if I was not searching for problems, I might be inclined to say the driver is working well in managed mode. Here is what I am continuing to see a lot of in the log:

```
[68572.277771] rtw89_8922au_git 4-3:1.0: failed to wait RF RX_DCK
[69028.100393] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69028.120383] rtw89_8922au_git 4-3:1.0: failed to wait RF PRE_NTFY
[69164.105168] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69421.106662] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69577.103586] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69577.123611] rtw89_8922au_git 4-3:1.0: failed to wait RF PRE_NTFY
[69577.187609] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69990.174051] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK
[69990.254007] rtw89_8922au_git 4-3:1.0: failed to wait RF TSSI
[70221.107576] rtw89_8922au_git 4-3:1.0: failed to wait RF DACK

```

Since my last report, I have moved to kernel 6.16. I was on kernel 6.12. I am using Debian 13 with KDE. With the move to kernel 6.16, iperf3 speeds seem to have smoothed out:

```
$ iperf3 -c 192.168.1.1
Connecting to host 192.168.1.1, port 5201
[  5] local 192.168.1.167 port 33430 connected to 192.168.1.1 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   143 MBytes  1.20 Gbits/sec    0   2.49 MBytes       
[  5]   1.00-2.00   sec   149 MBytes  1.25 Gbits/sec    0   2.49 MBytes       
[  5]   2.00-3.00   sec   146 MBytes  1.23 Gbits/sec    0   2.49 MBytes       
[  5]   3.00-4.00   sec   144 MBytes  1.20 Gbits/sec    0   2.49 MBytes       
[  5]   4.00-5.00   sec   163 MBytes  1.37 Gbits/sec    0   2.49 MBytes       
[  5]   5.00-6.00   sec   156 MBytes  1.31 Gbits/sec    0   2.49 MBytes       
[  5]   6.00-7.00   sec   142 MBytes  1.19 Gbits/sec    0   2.49 MBytes       
[  5]   7.00-8.00   sec   141 MBytes  1.18 Gbits/sec    0   2.49 MBytes       
[  5]   8.00-9.00   sec   139 MBytes  1.17 Gbits/sec    0   2.49 MBytes       
[  5]   9.00-10.00  sec   143 MBytes  1.20 Gbits/sec    0   2.49 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.43 GBytes  1.23 Gbits/sec    0            sender
[  5]   0.00-10.00  sec  1.43 GBytes  1.23 Gbits/sec                  receiver

```

I'll just continue searching and researching until I run onto something you can actually use.

morrownr, 2025-10-22:
I am back testing this rtw89_8922au_git driver. I need it to test some other things I am testing. I was responding to a couple of users reporting problems with AP with a mt7925e based card. I am using my mt7925u based adapter to try to duplicate the oriblems. So far the testing is going very well. I am running up to and including WiFi 6, 160 channel width, channel 65 (6 GHz). My only choice to test WiFi 7 is my rtl8912au based adapter. Has anyone tested WiFi 7 in managed mode with this driver yet?

@dubhater 

I do not have a WiFi 7 capable router yet and I am pondering if you do? If not, do we need to see about getting one to you? For myself, I am waiting on the OpenWRT Two to ship.

morrownr, 2025-11-05:
@dubhater 

I finally completed getting my RasPi4B/Netgear A9000 (WiFi 7) adapter going as a WiFi 7, tri-band AP so I have been testing rtw8922au for about a week now:

Band 4, 6 GHz
2 walls so less than optimal signal
WiFi 7
Hostapd: https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/hostapd-WiFi7.conf
hostapd is compiled with: https://github.com/morrownr/USB-WiFi/blob/main/home/AP_Mode/Upgrade_hostapd.md

```
$ iperf3 -c 192.168.1.24
Connecting to host 192.168.1.24, port 5201
[  5] local 192.168.1.237 port 47994 connected to 192.168.1.24 port 5201
[ ID] Interval           Transfer     Bitrate         Retr  Cwnd
[  5]   0.00-1.00   sec   104 MBytes   867 Mbits/sec    0   2.09 MBytes       
[  5]   1.00-2.00   sec   115 MBytes   962 Mbits/sec    0   2.09 MBytes       
[  5]   2.00-3.00   sec   113 MBytes   948 Mbits/sec    0   2.09 MBytes       
[  5]   3.00-4.00   sec   121 MBytes  1.01 Gbits/sec    0   2.09 MBytes       
[  5]   4.00-5.00   sec   118 MBytes   987 Mbits/sec    0   2.09 MBytes       
[  5]   5.00-6.00   sec   120 MBytes  1.01 Gbits/sec    0   2.09 MBytes       
[  5]   6.00-7.00   sec   120 MBytes  1.01 Gbits/sec    0   2.09 MBytes       
[  5]   7.00-8.00   sec   120 MBytes  1.00 Gbits/sec    0   2.09 MBytes       
[  5]   8.00-9.00   sec   120 MBytes  1.00 Gbits/sec    0   2.09 MBytes       
[  5]   9.00-10.00  sec   120 MBytes  1.01 Gbits/sec    0   2.09 MBytes       
- - - - - - - - - - - - - - - - - - - - - - - - -
[ ID] Interval           Transfer     Bitrate         Retr
[  5]   0.00-10.00  sec  1.15 GBytes   984 Mbits/sec    0            sender
[  5]   0.00-10.01  sec  1.14 GBytes   981 Mbits/sec                  receiver

```

```
$ sudo iw dev wlx7419f81676xx link
[sudo] password for morrow: 
Connected to 28:94:01:b4:e5:xx (on wlx7419f81676xx)
        SSID: myPI-WiFi7
        freq: 6295.0
        RX: 1130455526 bytes (2223249 packets)
        TX: 3440058411 bytes (673250 packets)
        signal: -58 dBm
        rx bitrate: 2161.3 MBit/s 160MHz EHT-MCS 10 EHT-NSS 2 EHT-GI 0
        tx bitrate: 2161.3 MBit/s 160MHz EHT-MCS 10 EHT-NSS 2 EHT-GI 0
        bss flags: short-slot-time
        dtim period: 2
        beacon int: 100

```

I have seen very stable Wifi 7 operation on band 4, 6 GHz, channel 69, for about a week on managed mode so things appear to be in good shape for managed mode on the rtw8922au driver.

Do you need me to test other things?

Also, a question regarding rtw89 in general: do any of the drivers in this repo or kernel have USB3 or LED support yet? I am seeing USB3 in my above testing with rtw8922au but I think it may be due to the previous driver I was using... things are not clear in this regard.

One last comment: The Netgear A9000, WiFi 7, mt7925 chip, is quite impressive in AP mode in case you or others can use this information. I have yet to test rtw8922au in AP mode as all of this requires a LOT of WORK and I am still in slow mode due to health.

dubhater, 2025-11-05:
> Do you need me to test other things?

I can't think of anything right now.

> Also, a question regarding rtw89 in general: do any of the drivers in this repo or kernel have USB3 or LED support yet? I am seeing USB3 in my above testing with rtw8922au but I think it may be due to the previous driver I was using... things are not clear in this regard.

Every driver in this repository has USB 3, except for RTL8851BU. I haven't seen any USB 3 devices with that chip.

The kernel doesn't have USB 3 support yet.

There is no LED support here or in the kernel. If the LED blinks it's because you loaded another driver before.

Do you have another driver for RTL8922AU? How is it?

dubhater, 2025-11-07:
> [@dubhater](https://github.com/dubhater)
> 
> > Every driver in this repository has USB 3, except for RTL8851BU. I haven't seen any USB 3 devices with that chip.
> 
> I have an adapter based on the rtl8851bu chip now but I am backed up on testing things. Given the specs on the adapter I would not think USB3 would be needed. Good to know that all the drivers here have USB3 support. I thought I was losing it. Do you have any idea when the USB3 support will go into the kernel?

It looks like next year.

> > There is no LED support here or in the kernel. If the LED blinks it's because you loaded another driver before.
> 
> Nothing supported in this repo is blinking.
> 
> > Do you have another driver for RTL8922AU? How is it?
> 
> I do not have another driver for the rtl8922au. I should have a copy if one is available but my sources have not indicated any availability. If I had the source I would get it to you or put it up here. Could it be that Realtek is not going to release an out-of-kernel driver for the rtl8922au? Possibly.

It exists... they just haven't finished it yet.

> Remember the code that went into the kernel to shut down WEXT and WiFi 7. Realtek had several years of notice that the code was in there... could this be what is causing the delay? My guess is that Realtek is THE REASON that the code was added to the kernel.

No, the code that disables WEXT for wifi 7 devices doesn't prevent them from making the usual style of drivers. They stopped relying on WEXT a while ago. They use cfg80211.