morrownr/rtw89 #29: 6GHZ Networks Not Visible

ev5unleash, 2025-09-16:
Hi,

I have installed these drivers on PopOS! 22.04 LTS Kernal 6.16.3-76061603-generic (Secure Boot Off) and was able to get the USB dongle ASUS Tri-Band WiFi 7 Nano USB Adapter (USB-BE92 Nano) to show up. However, it is unable to see any 6ghz-only networks, or connect on the 6ghz band on networks that support it. 

Any ideas why this might be? 

Thanks,

morrownr, 2025-10-15:
Hi @ev5unleash 

Has there been any progress on this issue?

dubhater, 2025-10-24:
@derpeter ``iw reg get`` is not supposed to show anything separate for this driver, just the "global" setting. What does it show?

To see what channels it's scanning:
```
echo 0x8000 > /sys/module/rtw89_core_git/parameters/debug_mask
```
Then trigger a scan and check dmesg.

morrownr, 2025-10-29:
Hi  @derpeter 

I am currently testing AP mode for an adapter with the mt7925u chip. One of the adapters that I am using as a client is based on the rtl8912au and I am using the driver driver here. What I have found with this rtl8912au based adapter is that I have to pull the adapter out of the USB port and put it back in before it will successfully work with 6 GHz. Sounds strange but...

pitt1717, 2026-09-09:
i just posted about the TXE70U..... i cant connect to 6ghz either. used chatgpt and i can see the freq but cant conect. using debian kde networkmanager, the wireless status shows its using wpa3, and it is set for US. i also added the US option to hard code into rtw89_8852cu.conf but it wont connect
where do i start?

a5a5aa555oo, 2026-09-10:
@pitt1717 What does `iw reg get` say?

pitt1717, 2026-09-10:
> [@pitt1717](https://github.com/pitt1717) What does `iw reg get` say?

global
country US: DFS-FCC
        (902 - 904 @ 2), (N/A, 30), (N/A)
        (904 - 920 @ 16), (N/A, 30), (N/A)
        (920 - 928 @ 8), (N/A, 30), (N/A)
        (2400 - 2472 @ 40), (N/A, 30), (N/A)
        (5150 - 5250 @ 80), (N/A, 23), (N/A), AUTO-BW
        (5250 - 5350 @ 80), (N/A, 24), (0 ms), DFS, AUTO-BW
        (5470 - 5730 @ 160), (N/A, 24), (0 ms), DFS
        (5730 - 5850 @ 80), (N/A, 30), (N/A), AUTO-BW
        (5850 - 5895 @ 40), (N/A, 27), (N/A), NO-OUTDOOR, AUTO-BW, PASSIVE-SCAN
        (5925 - 7125 @ 320), (N/A, 12), (N/A), NO-OUTDOOR, PASSIVE-SCAN
        (57240 - 71000 @ 2160), (N/A, 40), (N/A)

whats interesting is i added the following to the config. and while it complains of bad option lines, on occasion it will link to 6ghz. but not consistent at all. so not sure if its just a red herring 

cat /etc/modprobe.d/rtw89.conf

options 8852cu rtw_country_code=US   <-----------------added
options rtw89_core_git debug_mask=0x0
options rtw89_core_git disable_ps_mode=n

options rtw89_pci_git disable_clkreq=n
options rtw89_pci_git disable_aspm_l1=n
options rtw89_pci_git disable_aspm_l1ss=n

options rtw89_usb_git switch_usb_mode=y

options rtw_he_enable=y  <-----------------added (complains on modprobe)
options rtw_vht_enable=y <-----------------added (complains on modprobe)